dresden-elektronik/deconz-rest-plugin #770: Possibility to trigger mesh healing?

Kane610, 2018-09-03:
Is there any mechanic to be able to trigger the mesh to heal?

manup, 2018-09-03:
The mesh should heal on it's own. When routing fails new route discovery from the source node to the destination node will be started. This may take a while and prior to route discovery several attempts to send an request might be made.

Thats the theory :)

In practise meshing between different vendors has been challenging, the spec leaves some grey areas for interpretation so not all products work the same, mixing IKEA and Philips has caused troubles in the past. With firmware updates this will hopefully improve over time.

Kane610, 2018-09-03:
Thanks for educating me @manup!

closed 2018-09-03 by Kane610