dresden-elektronik/deconz-rest-plugin #778: xiaomi gateway support

clementx2006, 2018-09-08:
would it be possible to expose Xiaomi Gateway as a doorbell, siren and temperature / moisture sensor? i.e. a ZigBee router in the network with a few functions but not a hub...

rtenklooster, 2018-09-12:
Dont get it. The xiaomi gateway has an api so you can do whatever you like. 
The gateway doesn’t have a temperature / moisture sensor for as far as I know. 

What is it you want to accomplish?

clementx2006, 2018-09-14:
Apologies no temp/moisture sensor that's correct. 

It can play some ringtone/MP3 file
It has LED light
It can also broadcast internet radio 
It has zigbee chip

I was just wondering whether these functionalities can be exposed in DeCONZ. It's always plugged so it could be a great zigbee router in the zigbee network. Not as a hub but a zigbee router in the DeCONZ controlled network to begin with. Then a zigbee switch could turn on the internet radio, flash the light etc.

rtenklooster, 2018-09-14:
Just install node-red on your rpi.
You’re able to connect deconz to almost everything you can imagine. Node-red can listen to WebSocket events and you can control the gateway the way you like.

There’s even a node-red add on:
https://flows.nodered.org/node/node-red-contrib-mi-devices

Outlook voor iOS<https://aka.ms/o0ukef> downloaden

clementx2006, 2018-09-14:
okay many thanks for that - so it can't really be added to deconz as a zigbee router then?

closed 2019-01-20 by stale[bot]