dresden-elektronik/deconz-rest-plugin #779: Trådfri Smart Plug

MrPsi, 2018-09-10:
Hi,

Do you think the new Trådfri smart plug will work with deconz right away when it is released? Or does support have to be added?
https://www.ikea.com/se/sv/catalog/products/90356166/

Maybe hard to know before it has been released? :)

ebaauw, 2018-09-11:
Controlling the plug through the `/lights` resource will probably work without any changes.  Supporting the new on/off switch and any metering the plug might provide will require changes.

maxlein, 2018-09-16:
I just bought the new Outlet and it seems that it doesn't work with the current version. Gateway shows only my motion sensor. Is it configuration that needs to be changed or code?

ebaauw, 2018-09-16:
Could you post deCONZ GUI screenshots of:
- The node, incl. the endpoints and clusters;
- the _Node info_ panel;
- The _Cluster info_ panel of the _Basic_ cluster, **after** reading the attributes;

maxlein, 2018-09-16:
I am new to deconz, so I don't know if this is the right information:
Cluster info of the basic cluster I don't understand. I only see basic cluster info on the Osram plug.

[image]
[image]

maxlein, 2018-09-17:
No, I have a Osram plug and an Ikea. But the ikea one doesn‘t even show up. I think I tried the old and new web gui to search for devices, but nothing but the already connected devices were found.

manup, 2018-09-17:
Does it have a reset button or something like that?

Note: IKEA devices usually only try **once** to join the network.

* If they are factory new (which should be the case here)
* Directly after they are powered (light search must be started before that)

maxlein, 2018-09-17:
No, I couldn't find one. The motion sensor has a button but not the plug, at least not on the surface.
Will have a look at it again whn I am home today.

maxlein, 2018-09-17:
So the default app you have, gives me this screen:
[image]
Blink worked, but I can't see the plug anywhere but on this page.
The other app finds nothing:
[image]

manup, 2018-09-17:
Looks like the plug is already assigned to another network, so it needs to be reset. You may try using the red reset button in the old webapp, the plug needs to be very close to the gateway.

As alternative you can use a Hue dimmer switch to reset the plug, holding it very close to it and press buttons <key>I</key> and <key>O</key> for over 10 seconds.

zladukas, 2018-09-17:
@maxlein, 
manual says:

FACTORY RESET YOUR DEVICE For control outlet:
Push a pin into the pinhole on bottom of the control outlet for at least 5 seconds.

use it to pair device.

maxlein, 2018-09-17:
Thanks @zladukas, I threw the manual away and I could not find anything online!
I can see the plug now in decent gui. 
[image]

manup, 2018-09-17:
Cool, interesting, so it doesn't have power measurement but as a goody it seems to support ZigBee Green Power which is needed to route Hue tab and soon to be released ZGP switches.

Is it shown in the Phoscon App under lights?

ebaauw, 2018-09-17:
And, apparently, a different manufacturer, looking at the mac address.  (I should have done so earlier and recognised the first screenshot it was an OSRAM plug).

There's apparently an inconsistency in general.xml: the generic (ZigBee 3.0) device 0x010a is an `On/off plugin unit` without a `-`, the 0x0010 ZHA and ZLL devices are `On/off plug-in unit`.  Would this device ID be ZigBee 3.0 enough for the Hue bridge to support this plug?

> soon to be released ZGP switches

Interesting...

I think we still need deCONZ to configure the ZGP enabled routers or the ZGP devices will only work when in direct range of the RaspBee or ConBee, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/117.

manup, 2018-09-17:
>There's apparently an inconsistency in general.xml: the generic (ZigBee 3.0) device 0x010a is an On/off plugin unit without a -, the 0x0010 ZHA and ZLL devices are On/off plug-in unit. Would this device ID be ZigBee 3.0 enough for the Hue bridge to support this plug?

Not sure, maybe but we only know for sure after testing.

>I think we still need deCONZ to configure the ZGP enabled routers or the ZGP devices will only work when in direct range of the RaspBee or ConBee, see #117.

Absolutely, I sadly hadn't time yet to dig deeper in the ZGP spec, hopefully it's not too complicated to configure the ZGP proxies.

maxlein, 2018-09-18:
Yes, it shows in the Phsoscon app. Its also working in openhab through hue binding and in homekit. (y)

Filialen, 2018-09-21:
IKEA now has released the package including both the plug and the new on/off switch.
Here are the details of the switch…

[image]

[image]

sweebee, 2018-09-23:
Is the smart plug and remote already working? I cant get them paired. If it should work, how should i pair them?

manup, 2018-09-23:
>manual says:

>FACTORY RESET YOUR DEVICE For control outlet:
>Push a pin into the pinhole on bottom of the control outlet for at least 5 seconds.
>
>use it to pair device.

From a previous comment.

The remote isn't supported yet. See also [Supported Devices](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices#unsupported-switches)

sweebee, 2018-09-23:
Hm, i resetted it with the 5s method. Activated search for lights mode in deconz, but nothing shows up.

RudolfRendier, 2018-09-24:
Make sure the relay clicks (twice?) and the light on the device starts blinking.
For me it does show up when searching for lights.

manup, 2018-09-24:
Often, at least in case of the IKEA bulbs, the device needs to be power cycled once after the reset in order to start joining.

maxlein, 2018-09-24:
Well, I got the remote working. I just press the connect button a few times close to the antenna.
And with the plug, I just hold reset near the antenna, like described above from @zladukas

jurriaan, 2018-09-24:
> The remote isn't supported yet. See also Supported Devices

@manup What's the issue with these switches, can I help with something to get them supported?

robinjoh90, 2018-09-25:
> Well, I got the remote working. I just press the connect button a few times close to the antenna.
> And with the plug, I just hold reset near the antenna, like described above from @zladukas

@maxlein how did you add the remote? What type of switch did you choose when adding it?

maxlein, 2018-09-26:
Are you talking about the remote, or the switch for the plug? 
I am talking about the remote with the 5 diefferent buttons. And as I said, just pressing the rear connect button a few times near the antenna with an open gateway did the trick.

Filialen, 2018-09-26:
> I am talking about the remote with the 5 diefferent buttons. And as I said, just pressing the rear connect button a few times near the antenna with an open gateway did the trick.

We are talking about the small switch sold with the plug…
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/779#issuecomment-423656286
Of course, the remote control works. It has done that for a long time!

manup, 2018-11-24:
Closing this for now since the plug is supported, for the on/off remote the discussion is continued in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/853

closed 2018-11-24 by manup

mloebl, 2018-12-15:
Has anyone tried adding a smart plug recently?  I am running  2.05.50 and just got a couple of the smart plugs.  I did the reset thru the pin hole and they paired and showed up on my network, looking like the screen shots above in the deCONZ app.  I can also can control on/off them thru the deCONZ app.  However they are not showing up anywhere within the Phoscon app.  They do however seem to respond to All Lights off.

8engan, 2019-03-14:
For anyone else googling for this.
Do not give up, it took me a while but eventually I got it working and I think this was what made it work.
1. Add new light in phoscon
2. Reset plug (needle in bottom hole for 5 seconds, light starts cycling)
3. It showed up :)

Cavemanz, 2019-04-17:
I bought 2 of these today hoping to pair them with Conbee/Deconz. No matter what order I try to reset the plugs, nothing happens in the logs of the Deconz Docker. Ive tried more or less everything in this thread, and others. What am I missing?:(

Ive updated to the latest release of deconz and the latest firmware for Conbee.

manup, 2019-04-17:
I've paired one yesterday, here is what I did:

1. Start light search in Phoscon App
2. Press the button with a needle for approx. 10 seconds
3. Pull out the plug from power, reinsert after 5 seconds

It was then discovered quickly.

Cavemanz, 2019-04-17:
> I've paired one yesterday, here is what I did:
> 
> 1. Start light search in Phoscon App
> 2. Press the button with a needle for approx. 10 seconds
> 3. Pull out the plug from power, reinsert after 5 seconds
> 
> It was then discovered quickly.

How fast do you pull the plug on step 2? Straight after the "click"?

manup, 2019-04-18:
>How fast do you pull the plug on step 2? Straight after the "click"?

After a few seconds.

Do you have the ConBee connected directly to PC/RPi? If yes please try with an USB-extension cable.

Cavemanz, 2019-04-18:
Its connected through an USB extension already. Tried doing the include further away from the server, but no success. The plug only blinks for 3-4 seconds before going solid white. No matter when I reinsert the plug, it always is solid white afterwards.

masonevans, 2019-04-24:
@Cavemanz I had trouble resetting a couple of these.  It turns out I just wasn't pressing hard enough.  But if you're getting to the flashing then that doesn't seem like your issue.  Have you tried holding a pin in for longer?  The reset sequence is something like "click... flash a couple times... turn solid... fade out... flash again".  You need to hold until you get to that second flashing phase, then deCONZ should find it.

Cavemanz, 2019-04-24:
I do get the second flashing phase, but Deconz doesnt pick it up. I assume these logs from the docker aint related? 

19:25:10:405 Search sensors done,
19:24:22:385 Beacon src: 0xD5F3 ch: 11 updateId: 1,
19:24:21:873 Beacon src: 0xEBC2 ch: 11 updateId: 1,
19:24:21:857 Beacon src: 0x2A08 ch: 11 updateId: 1,
19:24:21:873 Beacon src: 0xD5F3 ch: 11 updateId: 1,
19:24:38:152 Beacon src: 0xD5F3 ch: 11 updateId: 1,
19:24:38:664 Beacon src: 0xEBC2 ch: 11 updateId: 1,
19:24:44:972 ZCL attribute report 0x8418260000DA2415 for cluster 0x0300, ep 0x03,
19:24:44:003 ZCL attribute report 0x8418260000DA2415 for cluster 0x0300, ep 0x03,
19:24:56:672 Beacon src: 0xEBC2 ch: 11 updateId: 1,
19:24:56:672 Beacon src: 0xD5F3 ch: 11 updateId: 1,
19:24:57:184 Beacon src: 0xEBC2 ch: 11 updateId: 1,
19:24:57:184 Beacon src: 0xD5F3 ch: 11 updateId: 1

manup, 2019-04-24:
The Beacons are a good sign that some device (the outlet?) tries to join.

One thing you can check: do you have by any chance multiple Phoscon App tabs open, perhaps on multiple devices? If so please close all tabs but the one in which the light search is running.

Cavemanz, 2019-04-24:
Sadly, I only had 1 window open on my main computer. Any other suggestions I could try? Getting a bit desperate. Id return the 2 I bought to IKEA but its a 1h drive so Id rather not if possible :)

manup, 2019-04-24:
Hmm strange, by the way which deCONZ and ConBee firmware are you running? Latest is deCONZ 2.05.64 and firmware version 0x26330500.

Cavemanz, 2019-04-24:
Version 2.05.63 / 3.4.2019
Firmware 26320500.
Updated to the newest firmware at the time, is there a fresh release now?
Ill grab 2.05.64 Docker image and try again.

manup, 2019-04-24:
Yes release is online, might be worth a try.

Cavemanz, 2019-04-24:
Anything in these logs that makes sense? Upgraded to newest version of Deconz and the firmware for Conbee.

https://pastebin.com/2FTYW84q

manup, 2019-04-24:
Log looks kinda ok, running out of ideas here.

Maybe try to power-off the the other mains powered devices while joining the plug? This should force it to join directly via ConBee.

Cavemanz, 2019-04-24:
Tried that now, no luck either. I dont get any Beacon log entries now either with my 4 lights switched off. I increased the search time to 3 min. It should show up as an actionable entry in the Lights section, right?

manup, 2019-04-24:
Yes should look like this:

[image]

Do you have access to the deCONZ GUI via VNC?

Only remaining thing to check are the Zigbee network parameters in the GUI.

Cavemanz, 2019-04-24:
Into the Docker container you mean? I havent tried that before.
[image]

manup, 2019-04-24:
That's the Phoscon App, showing only a limited set of the parameters.

The full Zigbee configuration is only visible in deCONZ GUI, with marthocs docker container it can be accessed via VNC (unfortunately I haven't done this myself yet) .

Configuration looks like

[image]

Cavemanz, 2019-04-24:
Managed to enable it in my docker compose file:
[image]

manup, 2019-04-24:
Hmm, these look fine too, all parameters are valid.
Very strange

Cavemanz, 2019-04-24:
I powered on my 4 lights again. Tried pairing both of my plugs. The Beacon src messages are back now, but nothing happens :/

https://pastebin.com/mMtZ01Ti

manup, 2019-04-24:
The beacons come from the lights, they are responses to the plug which requests them in order to find suitable routers to join.

Cavemanz, 2019-04-24:
Ok, Im close to giving up on these. Annoying that they wont pair seeing how Philips/Osram/Xiaomi pairs without issue :/

r41d, 2019-06-18:
A lot of times in this thread a hole in bottom of the plug is mentioned. Mine doesn't have one. How can I add it in deconz?

masonevans, 2019-06-19:
@r41d  if yours really doesn't have a hole in the bottom then you should return it because it's defective.  But make sure you're looking in the right place.  I would describe it as the "outside" more than the bottom, it's on the curved part of the plastic, on the side between the male part of the plug and the female part of the plug.  It's pretty small, about the size of a pin, but it's possible the plastic didn't get punched/molded properly and the hole is filled in.

r41d, 2019-06-19:
After taking pictures to show that I see no hole and uploading them I took another look and finally spotted the tiny hole at the bottom ^^. It's funny that you don't need it to for regular pairing with the official Gateway but that it's necessary for pairing with ConBee.
The pairing then worked flawlessly :)
What confused me was that the category was just called Light, it wasn't obvious to me that pairing of power plugs has to be done through this, as I also didn't really find any detailed instructions through Google. And I didn't notice the tiny hole, plus the normal pairing doesn't use it.

thowi, 2019-12-15:
It was really hard to pair mine. I tried multiple times to pair the socket, but failed many times.
At some point, I got saw "Range extender" by IKEA in Phoscon, but it was marked gray. I removed it, paired again, and eventually I saw it properly listed as "On/Off plug-in unit 13". Not sure why it's "13", maybe because I had to try 13 (?) times.
Glad it worked finally, but quite frustrating to the point that I thought I'd have to trash the device.

Does anybody own the IKEA gateway an know if pairing is also such a PITA there?
I.e., are the devices the problem, or can this be improved in Conbee/deCONZ?

ebaauw, 2019-12-15:
The IKEA hub works very differently: it only supports touchlink pairing.  You need to touchlink-pair a controller to the hub, and then touchlink-pair the controller to the light, plug, repeater, or window blind.  Note that the Trådfri motion sensors act as controllers.

austwhite, 2020-06-06:
I know this is closed, but curious if it got resolved?  Could it be the Zigbee channel being 11?  
I use channel 15, was considering buying a couple of these plugs, but I don't want to waste money if they are temperamental to install.