dresden-elektronik/deconz-rest-plugin #780: Firmwareupdate not possible under Ubuntu

Fl0ca, 2018-09-11:
Hi everyone,
i use deconz in an VM, hosted on an esxi Server. 

Since i switched from Raspberry to the VM deconz, it is not possible anymore, to read the Firmware from the Conbee as a result of that, i can not update the stick anymore.
The stick is working fine, only this is a little Problem.
Any Suggestions?

regards

olemr, 2018-09-11:
[Have a look here.](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/771#issuecomment-419558992)

manup, 2018-09-20:
Currently GCFFlasher in VMs and Docker don't work so well out of the box, it required to trigger a reset pin via USB control commands which don't pass through the virtual environment.

We are working on a rewrite of GCFFlasher which will address the issues, fix some bugs and  support VMs, Docker and macOs. It will take some time though and till it's available (hopefully in 2018) I recommend updating on a native platform.

ebaauw, 2018-09-20:
macOS!

manup, 2018-09-20:
:) yes!

closed 2019-01-25 by stale[bot]