dresden-elektronik/deconz-rest-plugin #781: deCONZ/hass.io/RaspBee Premium & FLS A lp or Hue not recognized

l-mb, 2018-09-11:
Dear all,

I just received my RaspBee Premium and the FLS A lp from Dresden Elektronik. I installed marthoc add-on fine, I get the login screen running.

But it does not recognize any lights during the commissioning step :-/ I tested the FLS A lp - the light after it is on, and the 0-10V outputs show 0.8V so it seems all working. I also bought a fresh-from-the-packaging Hue bulb to test with, and that one is also not found/discovered.

The red LED on the RaspBee is on, so that should indicate it's doing fine, right?

Any suggestions on how to debug this further would be welcome.

Thanks!

l-mb, 2018-09-12:
I've now been able to get my Hues to pair (using the official Dresden Elektronik firmware, there seems to be some annoying interactions with hass.io that I've not yet been able to resolve).

The FLS A lp is still not being recognized, even so. Running out of ideas ... Is there any way to test whether maybe I received a bad model?

manup, 2018-09-12:
Is the FLS shown in deCONZ GUI?

You can also try to reset the FLS with the following sequence:

[image]

l-mb, 2018-09-12:
Thanks for the response!

> Is the FLS shown in deCONZ GUI?

No, that's exactly what I'm not getting to work. (Even on the dedicated system running the firmware.)

> You can also try to reset the FLS with the following sequence:

I tried this sequence, but couldn't get the FLS A to reset. Several times. Of course, today I try it again _and_ it resets without any glitch. But it still doesn't pair :-(

(Also, this sequence has me worried. How would I do this if I need to re-pair in a completed installation, flip the fuse? ;-) )
  
A support engineer from Dresden Elektronik has now offered to swap the FLS A; since all other devices (lights, switches) pair, it seems not unlikely the FLS A might be problematic.

manup, 2018-09-12:
Can you provide more info on the used firmware and deCONZ version – it's visible in the Phoscon App > Menu > Settings > Gateway.

>Also, this sequence has me worried. How would I do this if I need to re-pair in a completed installation, flip the fuse? ;-)

There are 3 ways to reset a ballast:

1) Reset sequence
2) Touchlink reset from the old WebApp, requires the gateway/ConBee near the device
3) Delete via Phoscon App, this of course only works when the FLS is currently already connected connected 

>A support engineer from Dresden Elektronik has now offered to swap the FLS A; since all other devices (lights, switches) pair, it seems not unlikely the FLS A might be problematic.

Before that please check the versions as mentioned above, breaking the devices is very rare.

Also make sure that the on device Jumper configuration matches your setup and wiring.

l-mb, 2018-09-29:
I ended up resetting the RaspBee gateway entirely and starting over, including a factory reset via TouchLink - all processes done before, but that time it worked. So I'm calling it good ;-) Thanks.

closed 2018-09-29 by l-mb