dresden-elektronik/deconz-rest-plugin #782: Virtual light and hue motion

kitzler-walli, 2018-09-11:
I try to use Raspbee and a hue motion sensor to get motion information only an call shell scripts when motion occurs.
I was able to connect the hue motion sensor but the sensor simply does not report any "presence" at all.

Version 2.05.38 / 01/09/2018
Firmware 26230500

Here's the sensor output:
````
{
  "1": {
    "config": {
      "configured": true,
      "on": true,
      "sunriseoffset": 30,
      "sunsetoffset": -30
    },
    "etag": "14a2900bb5e3ab4c90316cf9defb8124",
    "manufacturername": "Philips",
    "modelid": "PHDL00",
    "name": "Daylight",
    "state": {
      "dark": true,
      "daylight": false,
      "lastupdated": "2018-09-11T17:24:29",
      "status": 200
    },
    "swversion": "1.0",
    "type": "Daylight",
    "uniqueid": "00:21:2e:ff:ff:02:47:6a-01"
  },
  "2": {
    "config": {
      "alert": "none",
      "battery": 100,
      "ledindication": false,
      "on": true,
      "pending": [],
      "reachable": true,
      "tholddark": 12000,
      "tholdoffset": 7000,
      "usertest": false
    },
    "ep": 2,
    "etag": "c422fc509f686a968fb3fd9072d0a513",
    "manufacturername": "Philips",
    "modelid": "SML001",
    "name": "Motion Sensor",
    "state": {
      "dark": true,
      "daylight": false,
      "lastupdated": "2018-09-11T17:30:51",
      "lightlevel": 8552,
      "lux": 7
    },
    "swversion": "6.1.0.18912",
    "type": "ZHALightLevel",
    "uniqueid": "00:17:88:01:03:29:3d:57-02-0400"
  },
  "3": {
    "config": {
      "alert": "none",
      "battery": 100,
      "delay": 0,
      "ledindication": false,
      "on": true,
      "pending": [],
      "reachable": true,
      "sensitivity": 0,
      "sensitivitymax": 2,
      "usertest": false
    },
    "ep": 2,
    "etag": "fc2a23e6907ad5af2ed5a9aa30329240",
    "manufacturername": "Philips",
    "modelid": "SML001",
    "name": "Motion Sensor",
    "state": {
      "lastupdated": "none",
      "presence": false
    },
    "swversion": "6.1.0.18912",
    "type": "ZHAPresence",
    "uniqueid": "00:17:88:01:03:29:3d:57-02-0406"
  },
  "4": {
    "config": {
      "alert": "none",
      "battery": 100,
      "ledindication": false,
      "offset": 0,
      "on": true,
      "pending": [],
      "reachable": true,
      "usertest": false
    },
    "ep": 2,
    "etag": "fc2a23e6907ad5af2ed5a9aa30329240",
    "manufacturername": "Philips",
    "modelid": "SML001",
    "name": "Motion Sensor",
    "state": {
      "lastupdated": "none",
      "temperature": null
    },
    "swversion": "6.1.0.18912",
    "type": "ZHATemperature",
    "uniqueid": "00:17:88:01:03:29:3d:57-02-0402"
  }
}
````

I already tried to reconnect and reset the motion sensor but that didn't solve the problem.
I assume there has to be a some setup for the sensor to report the presence information. 
Could anyone tell me how to set that up without any light configured? Maybe there is a way to add "virtual lights"?

Paalap, 2019-01-19:
Hi,

is there really no possibility to create a kind of a virtual light or bulb?
In my case I would like to use this virtual light to control my non-zigbee lights via iobroker (by forwarding the controls)

Thanks in advance

ebaauw, 2019-01-19:
You could create a CLIPGenericFlag sensor for `state.on` and a series of CLIPGenericStatus sensors for `state.bri`, `state.ct`, `state.hue`, `state.sat`, and with some fiddling `state.x` and `state.y`.  Personally, I wouldn't go there.  I use a CLIPGenericStatus to communicate the intent between home automation systems for better scaling and reduced complexity.

E.g. My deCONZ system determines whether a room is occupied, using the Hue motion sensors (in the room _and_ in adjacent rooms) with a series of deCONZ rooms.  This fact is stored in a CLIPGenericStatus sensor, which is exposed to HomeKit.  Using HomeKit automations, I turn on my Sonos speaker, switch them to radio input and set them to my preferred volume, when the room is occupied.  When I turn on my TV, the Sonos Playbar switches input.  HomeKit picks this up and sets the CLIPGenericStatus.  A deconz rule pick this up and activates the TV scene to dim the lights and soften their colour.

With this setup, HomeKit doesn't need to know the details of the lights in my room and deCONZ doesn't need to know the details of my Sonos setup.  Also this scales, as only a single message is needed to convey the intent (room occupied, watching tv), instead of an elaborate series of messages for each individual characteristic.