dresden-elektronik/deconz-rest-plugin #783: Xiaomi Door/Window sensor battery status

SwoopX, 2018-09-12:
Hey folks,

I'm trying to get this straight as I'm just about to integrate those sensors in my setup.

So apperantly, there's two versions available [https://community.smartthings.com/t/xiaomi-vs-aqara-door-window-sensors/90738](url) as per beginning of the thread and I got quite a bunch of the roundish ones. Some more of the other version are en route. From [https://github.com/dresden-elektronik/deconz-rest-plugin/issues/544](url) I see some people are in the same situation that battery does not seem to be reported.

I'd be interested in understanding what the reason might be. From my FHEM debug output, I can see that the battery value always returns "undef". However, since a friend has borrowed my dev gateway, I currently cannot dive into that further.

Hope this can be get working somehow as with 10+ sensors, this might be annoying :)

Maybe one thing off topic: I also got a FLS-PP lp. Can it be configured that it starts with an "off" setting, when it gets power connected?I have it running on a smart plug which gets toggled multiple times in the living room and having it "on" after power toggle is not too ideal.

manup, 2018-09-12:
> Maybe one thing off topic: I also got a FLS-PP lp. Can it be configured that it starts with an "off" setting, when it gets power connected?I

No currently this is not possible, the only workaround could be a REST-API rule that turns the light off after it gets powered (state/reachable).

SwoopX, 2018-09-12:
Thanks for coming back on this one so quickly, to bad...
Also thought about this workaround. So choices are either Phoscon, FHEM or use one of my Osram smart plugs. Guess I hve somthing to play around with on the weekend then.

apeinspace, 2018-11-10:
hi @manup, how is the roadmap with supporting the battery information from Xiaomi Sensors ? My sensors all have either been showing 100% or 88% for more than a year.

I know you are currently tackling bigger issues, but is this still planned to be implemented properly ? The only statement "it will be supported in the future" is from a year ago in #13 - hence the questions.

cheers

manup, 2018-11-10:
>hi @manup, how is the roadmap with supporting the battery information from Xiaomi Sensors ? My sensors all have either been showing 100% or 88% for more than a year.

The battery status in Xiaomi devices is supported since a while now, it's extracted from the hourly report command each Xiaomi sensor sends. So if you see 100% or 88% it is based on the values the device reports.

wvuyk, 2018-11-10:
I have a few Philips motion sensors that have been reporting 100% for 18 months. Both on Philips Hue bridge and deCONZ gateways. Last Februaryi finally some went down with their reports. I have one Philips motion sensor that is now reporting 0%, but still works fine after a 2 year+ of functioning with the same batteries. So far you are not yet on worlds best :-)

But so far Zigbee devices have been waaaaay more battery friendly than Z-Wave devices. I am guessing (also based on my Xiaomi sensors) patience might be a virtue.....

manup, 2019-01-28:
Closing this issue for now since the battery state is now available for Mija and Aqara devices.

closed 2019-01-28 by manup