dresden-elektronik/deconz-rest-plugin #785: Philips Hue dimmer switch lost after restart

smallgreyrock, 2018-09-13:
Version 2.05.38 / 9/1/2018
Firmware 26230500

Running deCONZ headless on an up-to-date Raspbian install.

When I reset a Philips Hue dimmer switch and add it to deCONZ, I am able to see button press events over the websocket. If I then restart deCONZ, the switch is still visible on the network, but I will never see button press events again.

Deleting, resetting, and re-adding the switch makes button press events show again, but only until the next restart.

Config files/DB are owned by the user running deCONZ, and the switches seem to be present in them.

smallgreyrock, 2018-09-13:
When I woke up this morning, button presses were registering again, but I decided to do a test.

This morning I deleted the database and re-added all lights plus one switch. After being added to deCONZ, button press events from that switch were sent to the websocket. I then restarted deCONZ, and no button press events were sent.

10 minutes later button press events still weren't being sent to the websocket.

I restarted again, this time using '--dbg-info=1 --dbg-aps=2 --dbg-zcl=1' and I can *immediately* see button presses in the console, but no events are being sent to the websocket.

I believe the switch is connected, deCONZ just isn't... doing anything with it. :)

smallgreyrock, 2018-09-14:
From a bit more experimenting:

I have a group of lights paired with the switch, and scenes configured for them in deCONZ.

When I restart deCONZ, paired switches work immediately. I am able to trigger scenes. When I run deCONZ with debug logging enabled, I see button presses being received by deCONZ. Events for the lights/scenes are sent to home assistant over the websocket, but button presses are not.

At some indeterminate point (more than 10 minutes, less than 8 hours after a restart) button events start getting broadcast on the websocket again.

ebaauw, 2018-09-15:
>  I see button presses being received by deCONZ

Are you sure?  The Hue dimmer switch sends different messages to control a group of lights directly (bypassing the deCONZ gateway) and to report buttonevents.  It sounds like the binding and/or attribute reporting for the buttonevents were missing after restart.

ebaauw, 2018-09-15:
So the first APS-DATA message (for cluster 0x0006) is the group command to switch the lights on, the second (for cluster 0xFC00) is the report for the buttonevent, a short release of the _On_ button.  This should generate a websocket event for `state.buttonevent`.  Is this an old log, or did you restart deCONZ after the buttonevents were sent to the websocket?

smallgreyrock, 2018-09-15:
> Is this an old log, or did you restart deCONZ after the buttonevents were sent to the websocket?

Neither, that's the issue.

That log is from immediately after restarting deCONZ. Button presses are being received, but they are **not** being sent out on the websocket.

At some unknown point after that (somewhere between 10 minutes and 8 hours), button presses start being sent over the websocket again.

I left it running overnight while I slept, and button presses are showing up on the websocket now.

ebaauw, 2018-09-15:
So now the buttonevents are sent over the websocket.  What happens if you restart deCONZ?  Do they continue to be sent, or does it take another 10 minutes to 8 hours before they're sent again?

smallgreyrock, 2018-09-15:
When I restart deCONZ it takes another 10 minutes to 8 hours before they're sent on the websocket again.

(but button presses will immediately register in the deCONZ log)

ebaauw, 2018-09-15:
That's very odd.  How do you check the web socket events?  Is your web socket client smart enough to re-open the socket when deCONZ restarts?

smallgreyrock, 2018-09-15:
Both from home-assistant (which reconnects) and a 5 line node.js client (which I restart)

Other events are visible, just not button presses.

smallgreyrock, 2018-09-15:
This is from the node.js websocket client before restarting, after deCONZ has been running for several hours:

```
Sat Sep 15 2018 22:08:10: {"e":"changed","id":"4","r":"sensors","state":{"lastupdated":"2018-09-15T22:08:10","presence":true},"t":"event"}
Sat Sep 15 2018 22:08:15: {"e":"changed","id":"7","r":"sensors","state":{"buttonevent":1000,"lastupdated":"2018-09-15T22:08:15"},"t":"event"}
Sat Sep 15 2018 22:08:15: {"e":"changed","id":"7","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2018-09-15T22:08:15"},"t":"event"}
Sat Sep 15 2018 22:08:15: {"e":"changed","id":"2","r":"lights","state":{"x":20414},"t":"event"}
Sat Sep 15 2018 22:08:15: {"e":"changed","id":"2","r":"lights","state":{"y":21478},"t":"event"}
Sat Sep 15 2018 22:08:15: {"e":"changed","id":"4","r":"lights","state":{"x":20414},"t":"event"}
Sat Sep 15 2018 22:08:15: {"e":"changed","id":"4","r":"lights","state":{"y":21478},"t":"event"}
Sat Sep 15 2018 22:08:15: {"e":"changed","id":"3","r":"lights","state":{"x":20414},"t":"event"}
Sat Sep 15 2018 22:08:15: {"e":"changed","id":"3","r":"lights","state":{"y":21478},"t":"event"}
Sat Sep 15 2018 22:08:16: {"e":"scene-called","gid":"1","r":"scenes","scid":"1","t":"event"}
```

This is immediately after restarting deCONZ and reconnecting the websocket client:

```
Sat Sep 15 2018 22:09:49: {"e":"changed","id":"1","r":"lights","state":{"colormode":"xy"},"t":"event"}
Sat Sep 15 2018 22:09:49: {"e":"changed","id":"1","r":"lights","state":{"x":20414},"t":"event"}
Sat Sep 15 2018 22:09:49: {"e":"changed","id":"1","r":"lights","state":{"y":21478},"t":"event"}
Sat Sep 15 2018 22:09:49: {"e":"changed","id":"2","r":"lights","state":{"x":20414},"t":"event"}
Sat Sep 15 2018 22:09:49: {"e":"changed","id":"2","r":"lights","state":{"y":21478},"t":"event"}
Sat Sep 15 2018 22:09:49: {"e":"changed","id":"4","r":"lights","state":{"x":20414},"t":"event"}
Sat Sep 15 2018 22:09:49: {"e":"changed","id":"4","r":"lights","state":{"y":21478},"t":"event"}
Sat Sep 15 2018 22:09:49: {"e":"changed","id":"3","r":"lights","state":{"x":20414},"t":"event"}
Sat Sep 15 2018 22:09:49: {"e":"changed","id":"3","r":"lights","state":{"y":21478},"t":"event"}
Sat Sep 15 2018 22:09:50: {"e":"scene-called","gid":"1","r":"scenes","scid":"1","t":"event"}
```

ebaauw, 2018-09-16:
Maybe related to 
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/615#issuecomment-421657336?  I see `lastpdated` has the same value as the log timestamp, but the latter should be in local time and `lastupdated` in UTC.

smallgreyrock, 2018-09-16:
Yep, that seems to have fixed it.

Do they need to match? Is there any way to have deCONZ just use the system tz?

ebaauw, 2018-09-16:
Afaik, there’s no way for deCONZ to copy the system timezone.  It keeps its own setting in `/config/timezone`.  You can use the old web app to change it (or the API); there’s no support for that in Phoscon.

@manup, would it be possible also to set the timezone, when automatically setting the latitute / longitude for the daylight sensor?

manup, 2018-09-16:
>@manup, would it be possible also to set the timezone, when automatically setting the latitute / longitude for the daylight sensor?

There is already code that does fiddling with timezone detection, but the result is these GMT+2 settings. 

However meanwhile there are better JS interfaces to detect timezone, language and 24/12 h hour format available in modern browsers, these can yield the correct settings like Europe/Paris. I'll put it on the todo list, we need to refactor the Phoscon App configuration code anyway so this should be part of it.

closed 2019-01-21 by stale[bot]

ottelo9, 2019-04-22:
I have the same problems with my 3 Hue Dimmer Switches. All 3 switches have no connection linies anymore. If I press a button the blue circle flashes but nothing happens.
[image]