dresden-elektronik/deconz-rest-plugin #789: Philips Hue Motion displayed in deCONZ but not in rest-plugin

ma-ca, 2018-09-16:
Philips Hue Motions joins the network and is displayed in deCONZ but not in rest-plugin.

Using Raspbee with deCONZ 2.05.38 / firmware 0x26230500.

I removed the battery, clicked Open Network in old web app, and inserted battery again. The Hue Motion joins the network as seen in the log, but not visible in rest-plugin.

Then I again opened network, pushed setup buttion on Hue Motion and again the Hue Motion joins the network. The logs shows a new NWK addr 0x69D5. But still not visible in rest-plugin `/api/xxx/sensors`.

What else is needed for Hue Motion to show up in the rest-plugin?

```
deCONZ --http-port=80 --dbg-info=1 --dbg-aps=2 --dbg-zcl=1 -platform minimal >  deconz.log

grep -e 0x66D5 -e 0x0017880102035825 -e 0x69D5 -e "permit join" -A 1 deconz.log

10:57:16:474 Node: id: 2, 00:17:88:01:02:03:58:25 (0x0017880102035825) scene: -122.000000, -68.000000
10:57:16:526 Node: id: 3, 00:1f:ee:00:00:00:17:0a (0x001FEE000000170A) scene: 1.000000, -58.000000
--
11:00:19:204 APS-DATA.indication srcAddr: 0x0017880102035825, dstAddrMode: 2, profile: 0x0000, cluster: 0x0006, lqi: 255, rssi: -38
11:00:19:204 	asdu: 9bfdff040101190000
--
11:00:20:751 APS-DATA.indication srcAddr: 0x0017880102035825, dstAddrMode: 2, profile: 0x0000, cluster: 0x0001, lqi: 255, rssi: -37
11:00:20:751 	asdu: 9c00000000
--
11:00:21:333 APS-DATA.indication srcAddr: 0x0017880102035825, dstAddrMode: 2, profile: 0x0104, cluster: 0x0019, lqi: 255, rssi: -37
11:00:21:333 	asdu: 017701000b100d01e0490042
--
11:02:17:276 send permit join, duration: 59
11:02:17:469 APS-DATA.confirm id: 216, status: 0x00 SUCCESS
--
11:02:22:730 APS-DATA.indication srcAddr: 0x0017880102035825, dstAddrMode: 2, profile: 0x0000, cluster: 0x0013, lqi: 255, rssi: -39
11:02:22:730 	asdu: 00d566255803020188170080
11:02:22:730 device announce 0x0017880102035825 (0x66D5) mac capabilities 0x80
11:02:22:730 set fast probe address to 0x0017880102035825 (0x66D5)
11:02:22:731 device announce 0x0017880102035825 (0x66D5) mac capabilities 0x80
11:02:22:731 set fast probe address to 0x0017880102035825 (0x66D5)
11:02:23:775 [2] get active endpoints for 0x0017880102035825
11:02:23:775 APS-DATA.request id: 252, addrmode: 0x02, addr: 0x66d5, profile: 0x0000, cluster: 0x0005, ep: 0x00 -> 0x00 queue: 0 len: 3 tx.options 0x00
--
11:02:24:337 APS-DATA.indication srcAddr: 0x0017880102035825, dstAddrMode: 2, profile: 0x0000, cluster: 0x8005, lqi: 255, rssi: -37
11:02:24:337 	asdu: 4700d566020201
--
11:02:24:443 [4] get basic cluster attr 0x0004 for 0x0017880102035825
11:02:24:443 APS-DATA.request id: 1, addrmode: 0x02, addr: 0x66d5, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x02 queue: 0 len: 5 tx.options 0x00
--
11:02:24:876 APS-DATA.indication srcAddr: 0x0017880102035825, dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 255, rssi: -37
11:02:24:877 	asdu: 18480104000042075068696c697073
11:02:24:884 [4] get basic cluster attr 0x0004 for 0x0017880102035825
11:02:24:884 APS-DATA.request id: 6, addrmode: 0x02, addr: 0x66d5, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x02 queue: 0 len: 5 tx.options 0x00
--
11:02:25:380 APS-DATA.indication srcAddr: 0x0017880102035825, dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 255, rssi: -37
11:02:25:380 	asdu: 18490104000042075068696c697073
11:02:25:396 [4] get basic cluster attr 0x0004 for 0x0017880102035825
11:02:25:396 APS-DATA.request id: 13, addrmode: 0x02, addr: 0x66d5, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x02 queue: 0 len: 5 tx.options 0x00
--

...
(same message repeated many many times)

--
11:02:41:752 APS-DATA.indication srcAddr: 0x0017880102035825, dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 255, rssi: -40
11:02:41:752 	asdu: 18690104000042075068696c697073
11:02:41:758 [4] get basic cluster attr 0x0004 for 0x0017880102035825
11:02:41:758 APS-DATA.request id: 193, addrmode: 0x02, addr: 0x66d5, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x02 queue: 0 len: 5 tx.options 0x00
--
11:02:42:263 APS-DATA.indication srcAddr: 0x0017880102035825, dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 255, rssi: -40
11:02:42:263 	asdu: 186a0104000042075068696c697073
11:02:42:269 [4] get basic cluster attr 0x0004 for 0x0017880102035825
11:02:42:269 APS-DATA.request id: 198, addrmode: 0x02, addr: 0x66d5, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x02 queue: 0 len: 5 tx.options 0x00
--
11:04:21:291     * different nwk address 0x66D5 / 0x69D5
11:04:21:619 APS-DATA.request id: 190, addrmode: 0x03, addr: 0x001fee0000001739, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
--
11:06:17:931     * different nwk address 0x66D5 / 0x69D5
11:06:18:307 APS-DATA.request id: 72, addrmode: 0x03, addr: 0x001fee0000001739, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
--
11:08:14:184     * different nwk address 0x66D5 / 0x69D5
11:08:14:279 poll node 00:1f:ee:00:00:00:24:ce-03-0b04
--
11:10:10:530     * different nwk address 0x66D5 / 0x69D5
11:10:10:775 DB save zll database items 0x00000880
--
11:11:57:291     * different nwk address 0x66D5 / 0x69D5
11:11:57:350 APS-DATA.confirm id: 202, status: 0x00 SUCCESS
--
11:13:42:997     * different nwk address 0x66D5 / 0x69D5
11:13:43:372 APS-DATA.request id: 39, addrmode: 0x03, addr: 0x001fee0000001739, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
--
11:13:52:277 send permit join, duration: 59
11:13:52:478 APS-DATA.confirm id: 84, status: 0x00 SUCCESS
--
11:13:57:915 APS-DATA.indication srcAddr: 0x0017880102035825, dstAddrMode: 2, profile: 0x0000, cluster: 0x0013, lqi: 207, rssi: -65
11:13:57:915 	asdu: 00d569255803020188170080
11:13:57:915 device announce 0x0017880102035825 (0x69D5) mac capabilities 0x80
11:13:57:915 set fast probe address to 0x0017880102035825 (0x69D5)
11:13:57:916 device announce 0x0017880102035825 (0x69D5) mac capabilities 0x80
11:13:57:916 set fast probe address to 0x0017880102035825 (0x69D5)
11:13:57:916 0x66D5 nwk changed to 0x69D5
11:13:58:280 poll node 00:1f:ee:00:00:00:17:0a-03-0006
--
11:13:58:865 [2] get active endpoints for 0x0017880102035825
11:13:58:866 APS-DATA.request id: 124, addrmode: 0x02, addr: 0x69d5, profile: 0x0000, cluster: 0x0005, ep: 0x00 -> 0x00 queue: 0 len: 3 tx.options 0x00
--
11:13:59:481 APS-DATA.indication srcAddr: 0x0017880102035825, dstAddrMode: 2, profile: 0x0000, cluster: 0x8005, lqi: 215, rssi: -64
11:13:59:481 	asdu: 2300d569020201
--
11:13:59:577 [4] get basic cluster attr 0x0004 for 0x0017880102035825
11:13:59:577 APS-DATA.request id: 129, addrmode: 0x02, addr: 0x69d5, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x02 queue: 0 len: 5 tx.options 0x00
--
11:13:59:995 APS-DATA.indication srcAddr: 0x0017880102035825, dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 199, rssi: -66
11:13:59:995 	asdu: 18240104000042075068696c697073
11:14:00:001 [4] get basic cluster attr 0x0004 for 0x0017880102035825
11:14:00:001 APS-DATA.request id: 133, addrmode: 0x02, addr: 0x69d5, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x02 queue: 0 len: 5 tx.options 0x00
--
11:14:00:517 APS-DATA.indication srcAddr: 0x0017880102035825, dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 207, rssi: -65
11:14:00:517 	asdu: 18250104000042075068696c697073
11:14:00:524 [4] get basic cluster attr 0x0004 for 0x0017880102035825
11:14:00:525 APS-DATA.request id: 138, addrmode: 0x02, addr: 0x69d5, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x02 queue: 0 len: 5 tx.options 0x00
--
11:14:01:027 APS-DATA.indication srcAddr: 0x0017880102035825, dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 199, rssi: -66
11:14:01:027 	asdu: 18260104000042075068696c697073
11:14:01:034 [4] get basic cluster attr 0x0004 for 0x0017880102035825
11:14:01:034 APS-DATA.request id: 144, addrmode: 0x02, addr: 0x69d5, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x02 queue: 0 len: 5 tx.options 0x00
--
11:14:17:419 [4] get basic cluster attr 0x0004 for 0x0017880102035825
11:14:17:420 APS-DATA.request id: 226, addrmode: 0x02, addr: 0x69d5, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x02 queue: 0 len: 5 tx.options 0x00
--

(same message repeated many many times)

```
[image]

---

[image]

---

[image]

ebaauw, 2018-09-16:
Read the attributes from the _Basic_ cluster in the _Cluster info_ panel in the GUI, while the network was opened from the old web app.

ma-ca, 2018-09-16:
Thanks, that solved it.

closed 2018-09-16 by ma-ca