dresden-elektronik/deconz-rest-plugin #791: UDP Port 1900 is used , can this be changed ?

SkinnyNorris123, 2018-09-19:
Hi,
on my fhem installation where deCONZ works in parallel to SONOS I notice that
they can not really live site-by-site :-(
SONOS-Plugin needs to create the SSDP-multicast listener socket on UDP-Port 1900.
Unfortunate deCONZ uses this port already.
`
pi@raspberrypi:/opt/fhem/log $ sudo netstat -a | grep 1900
udp        0      0 *:1900                  *:*
pi@raspberrypi:/opt/fhem/log $ sudo netstat -apn | grep 1900
udp        0      0 0.0.0.0:1900            0.0.0.0:*                           364/deCONZ
`
Is there a way to start deCONZ without this, or to use an different port for it ?

manup, 2018-09-19:
Yes you can start deCONZ with commandline parameter `--upnp=0` to disable SSDP/UPnP.

SkinnyNorris123, 2018-09-19:
Thank you very much for this VERY fast perfect answer. 
But, I`m not a linux expert, how do I modify this ?
I`m starting deCONZ automatically like explained here 👍 [https://github.com/dresden-elektronik/deconz-rest-plugin](url)

SkinnyNorris123, 2018-09-19:
What I did is 
vi /usr/lib/systemd/system/deconz.service 
and add --upnp=0
`
[Service]
User=1000
ExecStart=/usr/bin/deCONZ -platform minimal --http-port=80 **--upnp=0**
Restart=on-failure
StartLimitInterval=60
AmbientCapabilities=CAP_NET_BIND_SERVICE CAP_KILL CAP_SYS_BOOT CAP_SYS_TIME

`

manup, 2018-09-19:
If you use the headless version then this should work, for the gui version you would need to modify deconz-gui.service.

Be aware that changes to the files in /usr/lib/systemd/system/deconz.service wilol be lost after an deCONZ update, you may use systemd overwrite service files instead.

SkinnyNorris123, 2018-09-19:
Vielen Dank :-)

manup, 2018-09-20:
Hope it works, close the issue for now. :)

closed 2018-09-20 by manup