dresden-elektronik/deconz-rest-plugin #792: Smartenit ZBMLC30 ??

dem5867, 2018-09-19:
What information would I need to get to you so you can add this device.. It presently shows up as a light {I'm a NOOB so it maybe just installed incorrectly } and it does switch on and off , but I beleave it can also supply power readings... Thank You.

manup, 2018-09-19:
Best information are provided by screenshots from the node endpoints and *Cluster Info Panel* of basic cluster after reading its attributes.

Also Screenshot of *Node Info Panel*.

dem5867, 2018-09-19:
Is the endpoint the data under the word dryer ?
[image]
[image]
[image]
I don't think the following EndPoint information is correct , how do I get the correct endpoint information ??
[image]

dem5867, 2018-09-19:
I don't know if this helps , yet could it be I did not add it correctly. I just held the button until it was blinking and it got added as the device above ,, yet it appears one has to join and bind the device... see below..
[image]

manup, 2018-09-19:
Thanks for the details, with the above screenshots we can add the power readings via Simple Metering cluster or at least start to whitelist them.

ebaauw, 2018-09-22:
The _Simple Metering_ cluster usually only reports life-time consumption (kWh), and sometimes current power (W).  Try reading the cluster in the deCONZ GUI: _Current Summation Delivered_ (0x0000) is the life-time consumption; _Instanteneous demand_ (0x0400) the current power.  The scales typically differs per device, and only a few devices support the formatting attributes.

For voltage (V), current (A), and sometimes frequency (Hz), you'd need a _Electrical Measurement_ cluster (0x0B04).  This cluster also contains power (W).

closed 2019-01-27 by stale[bot]