dresden-elektronik/deconz-rest-plugin #794: Device support request: Orvibo in-wall dimmer switch

udoewich, 2018-09-22:
I took a chance and bought an Orvibo in-wall Zigbee dimmer switch (US version; box says model# T10D1) from a local reseller to see if I could get it to work with Phoscon.  Alas, it does not show up with either a light or sensor scan.  It did however pair with my Conbee and shows up in deCONZ.  Here's the info:

[image]

Basic Cluster Info:

[image]

Basic Node Info:

[image]

Would love to see this added as a light (or switch?) in Phoscon.  I was already able to turn the connected lights on, off and dim them via deConz.

manup, 2018-09-22:
Link to the Orvibo product line https://www.orvibo.com/en/product/productList.html

udoewich, 2018-09-24:
I wasn't sure if off-site links are allowed on GitHub, but yes, that's the manufacturers web site.  I'm holding off on getting more of their switches until I know I can control them via Phoscon.  I also have my eye on their simple on/off in-wall switch.

Is there a limit of how many such devices can be controlled by a Conbee?

manup, 2018-09-25:
>Is there a limit of how many such devices can be controlled by a Conbee?

Officially we say 200 but there isn't really a hard limit, depends on the setup so more might be possible.

udoewich, 2018-11-27:
Does it make sense for me to upgrade to 2.05.49, or is there no support for this device yet?  I'm still holding off buying more of these switches until I know I can remote control them via Phoscon...

Thanks in advance.

PR #1763 (2019-08-26): Support for ORVIBO switch, dimmer and motion sensor
This adds support for ORVIBO in-wall switch, in-wall dimmer and motion sensor. Closes #794, #1686 and #1747.
closed 2019-04-04 by stale[bot]