dresden-elektronik/deconz-rest-plugin #796: Firmware update blues

ebaauw, 2018-09-23:
There's still an issue updating the firmware under v2.05.39.  When both a RaspBee and ConBee are connected, firmware upgrade (from the web app) still updates the ConBee, even though deCONZ uses the RaspBee.  From the log:
```
Sep 23 11:14:28 pi1 sudo[2360]:       pi : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/GCFFlasher_internal -sn DJ00Q8ZR -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26240500.bin.GCF
```

After disconnecting the ConBee, it works:
```
Sep 23 11:21:58 pi1 sudo[4929]:       pi : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/GCFFlasher_internal -d RaspBee -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26240500.bin.GCF
```

There's also a bug in ZShark v1.01 causing the ConBee firmware upgrade to the sniffer firmware to fail:
```
$ zshark 
libEGL warning: DRI2: failed to authenticate
libpng warning: iCCP: known incorrect sRGB profile
loeth011:31:17:370 Connected to /dev/ttyUSB0
11:31:17:657 Disconnected
11:31:17:659 Connected to /dev/ttyUSB0
11:31:17:951 Disconnected
"GCFFlasher V2_11 (c) dresden elektronik ingenieurtechnik gmbh 2017/12/10
using firmware file: /usr/share/zshark/fwsniffer_v1_0.bin 
error opening firmware file /usr/share/zshark/fwsniffer_v1_0.bin
"
```
There's a `/` missing in the firmware filename.  As a workaround, create a symlink for the file that zshark is expecting.
```
$ ln -s /usr/share/zshark/fw/sniffer_v1_0.bin /usr/share/zshark/fwsniffer_v1_0.bin 
```

wvuyk, 2018-09-24:
I also have issues with the firmware upgrades. I have a ConBee and a RaspBee. Ever since 2.0.4.00 or so my ConBee is reposting an old version firmware 0x261F10500. With every firmware upgrade I press the upgrade button in  Phoscon. Earlier Phoscon would report versio 0x00000 after the upgrade, but deCONZ still reports 0x261F10500.
The RaspBee seems to update normally.

How can I get the ConBee to really update the firmware?

manup, 2018-09-24:
Do you use Windows 10?

I've only testet the update with Windows 7 (user has admin rights) which always worked, maybe there is a difference, need to check it in my Win 10 VM and also with non admin rights.

wvuyk, 2018-09-24:
I am using Windows 10 here. I installed the software on a different location (not the users directory) so I can make an easy backup of it when needed. Maybe this different location (e:\deCONZ) is causing issues? I discovered that the firmware folder was holding 4 files on the changed installation dir (e:\deCONZ\Firmware), but it was also holding a duplicate in the users appdata\Local directory?

For testing I installed deCONZ as default in the users directory on a different machine and updated my ConBee firmware with this installation. This works immediatly correct... so I am thinking deCONZ does not like being installed in a different location then the users directories? Which I think is not recommendable to do?

ebaauw, 2018-11-27:
ZShark v1.02 finds the correct firmware file.

deCONZ v2.05.49 seems to update the RaspBee even when a ConBee is connected.

closed 2018-11-27 by ebaauw

Elliott32224, 2019-02-18:
I am trying to update Conbee deConz firmware to 262F0500 via Phoscon Version 2.05.59 / 2/14/2019 on Windows 10. It goes through the process and I can see the progress of the updating. But, when I go back to Phoscon, it shows "a new firmware is available." I read somewhere to remove other devices from USB ports on the computer, so that the Conbee was the only device connected. This did not make a difference. Any other suggestions? Thanks.

manup, 2019-02-18:
You may try to run the firmware update manually, this also provides more insight about errors.

* Close deCONZ
* Download GCFFlasher for Windows:
https://www.dresden-elektronik.de/funktechnik/service/downloads/software/
* Unpack the zip
* Download the Firmware and save to the GCFlasher directory
https://www.dresden-elektronik.de/rpi/deconz-firmware/deCONZ_Rpi_0x262f0500.bin.GCF
* Double click `GCFFlasherCommandline.bat`
* Run the update manually in the opened command prompt

    `GCFFlasher -d 0 -f deCONZ_Rpi_0x262f0500.bin.GCF`

artkrv, 2021-04-19:
I have used @manup sollution. Mine was flapping all the time. 
On windows:
`.\GCFFlasher -d COM8 -t 60 -f ..\firmware\deCONZ_ConBeeII_0x264a0700.bin.GCF`