dresden-elektronik/deconz-rest-plugin #800: deConz GUI over remote PC

thrusty66, 2018-09-23:
Hello, I have deConz running on an RPi 3, in conjunction with the RaspBee Shield. The Raspberry runs completely headless over SSH. I installed Raspbian light.  
Now my problem is how to call deConz GUI. Is it possible to run the GUI on a remote Ubuntu PC?
 When I call the IP of the Raspberry via a browser, I only get a page where I can connect lamps. There are no further settings. There is no possibility to learn a sensor. 
Maybe I'm just making a rookie mistake. Anyway, I'm not getting anywhere at the moment. 
Or do I have to learn a light source before I get further options?
Currently I don't have a ZigBee lamp, only an Aqara PirSensor. Therefore I cannot register a lamp at the moment.

manup, 2018-09-23:
>Or do I have to learn a light source before I get further options?

Yeah that was the initial assumption for the app :) we'll later add a button to skip this.

Meanwhile however to get over this page you can simply replace **commissioning.html** with **index.html** in the URL bar of your browser.

thrusty66, 2018-09-23:
Thanks for the quick answer. Now I can search for sensors. Mine is not connecting at the moment. But there could be other reasons.

manup, 2018-09-23:
Make sure to use the latest version as described in https://github.com/dresden-elektronik/deconz-rest-plugin#installation

And since this a new installation you will likely need to update the RaspBee firmware in *Menu > Gateway > Settings* (current firmware is version 0x26240500).

thrusty66, 2018-09-23:
I tried to update the firmware via the WebGUI. But I'm not sure if it worked. 
The GUI shows me the following:

Manufacturer dresden elektronik
Product deCONZ
Version 2.05.23 / 25.4.2018
Firmware 00000000
The version is up to date.

bdickhaus, 2018-09-24:
This happens to me too. Restarting deconz from the command line gets it back.

manup, 2018-09-24:
The firmware being 00000000 problem is addressed in 2.05.24, however there are still some odd ends with browser cache which we'll address in the next releases.

thrusty66, 2018-09-24:
I'm afraid my problems have a completely different nature. Today I went to IKEA and got one of these lamps. But even this one does not let me register. 
Is there a HowTo somewhere? A HowTo for beginners. 
At the moment I have problems to understand the connections. I think the firmware on the RasBee has to be loaded independently from the Raspberry software. Is that correct? The RaspBee tried to load a new firmware a few times yesterday. Apparently without success. Currently the RaspBee is not trying to load a new firmware. Only the message: Firmware current. 
Do I need qt5 on the RaspBee? How can I change the firmware of the RaspBee "manually"? 
Sorry, but how do I get the firmware 2.05.24 on the RaspBee?

SwoopX, 2018-09-25:
Hey there,

I may give you a hand on this. As I see it, some command line magic would do the trick. Had the issue myself deconz not updating the firmware some time ago, probably due to the very first firmware running on it.

If it works out, I'm happy to contribute a couple of lines to the wiki later on. However, I'll leave it in english, so everyone can participate 🙂

SwoopX, 2018-09-25:
[https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually](url)

SwoopX, 2018-09-26:
Hm, at least in my case, the red light on the RaspBee is on permanently and the little green one only during command transmission. Maybe a screwed setup? Are you running in headless mode or with a Xserver? What does "sudo systemctl status deconz" give back?

thrusty66, 2018-09-26:
I don't understand what you mean "Maybe a screwed setup"? The RPi is currently not in any case, just a test setup on the desk. 
I get this issue:
```
pi@raspberrypi:~` $ sudo systemctl status deconz
● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2018-09-25 17:17:03 UTC; 21h ago
 Main PID: 324 (deCONZ)
   CGroup: /system.slice/deconz.service
           └─324 /usr/bin/deCONZ -platform minimal --http-port=80

Sep 25 17:17:03 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
Sep 25 17:17:05 raspberrypi deCONZ[324]: libpng warning: iCCP: known incorrect sRGB profile
Sep 25 17:17:06 raspberrypi deCONZ[324]: This plugin does not support propagateSizeHints()
Sep 25 17:17:06 raspberrypi deCONZ[324]: This plugin does not support propagateSizeHints()
```

manup, 2018-09-26:
It looks like deCONZ hasn't a connection to the serial port of RaspBee.

Please check if the serial port is configured as described here:
https://github.com/dresden-elektronik/deconz-rest-plugin#software-requirements

I still wonder why no red led is on, which would mean that the ZigBee network isn't up.

thrusty66, 2018-09-26:
I tried several times to activate the serial via Raspi-config. Is there any way I can test if it worked? 

[On this page](
https://www.dresden-elektronik.de/fileadmin/Downloads/Dokumente/Produkte/ZLL/RaspBee-BHB-en.pdf#page12) there is a description of the functions of the LEDs. But I haven't gotten any further with it either.

thrusty66, 2018-09-26:
Is it possible to upload a picture here? Or do I have to save it somewhere else and link it here?

manup, 2018-09-26:
Yes, just copy it and paste it here in the editor, that's the easiest way.

thrusty66, 2018-09-26:
Thanks, it works.
This is my WebGUI before and after the alleged firmware update. 

[image]
[image]
[image]
[image]

thrusty66, 2018-09-26:
I suspect the deCONZ software has no connection to the RaspBee. Where does the display of a firmware version come from? And after the update it shows the older firmware version? Maybe deCONZ loaded the firmware and just didn't transfer it to RasBee. 
Something is totally wrong, but I don't know what.

manup, 2018-09-26:
Looks really weird especially the older version of deCONZ after update, might also be a browser cache issue. You can force force refresh of the cache in the URL bar by adding bogus query string like `http://192.168.178.72/pwa/settings-gateway.html?123`

However if it still says 'not connected / nicht verbunden' it might be best to temporary use the official sd-card image and check if that works out of the box (as it should), otherwise the ZigBee configuration must be verified within deCONZ settings.

SwoopX, 2018-09-26:
I recall a quite similar behavior when I first tried to run deconz under a different user account as "pi". Deconz was running but I wasn't able to get any connections.

It's almost certain some prerequisites are not given. Do you still have the user "pi" and is it member of the group "dialout"?

thrusty66, 2018-09-26:
I just kind of brought the green LED back to life. Somehow I managed to update the WebGUI again. And the green LED on the RaspBee known to light up again. According to the description, the green LED lights up when data is transferred!?
Then the update was finished and the green LED continues to guide you now no longer goes out.
The WebGUI still shows me that there is no connection to the firmware.
After a soft restart and also after a hard restart the LED stays off.

manup, 2018-09-26:
>According to the description, the green LED lights up when data is transferred!?

Yes during firmware flashing it blinks for up to one minute.

> Only the red LED I get no longer moved to glow.

So all leds are off?

Is this with the official sd-card image or the headless setup?

SwoopX, 2018-09-26:
Hm, not sure if that would make any difference but you could try to disable Bluetooth, just to be sure [https://forum.smartapfel.de/forum/thread/479-installation-raspbee-von-dresden/?pageNo=24](url)

You also might want to check if the RaspBee devices gets hot. Had that a couple of weeks ago due to my stupidity attaching to to another raspi.  Got somehow broken, may it rest in peaces...

thrusty66, 2018-09-27:
Yes, all LEDs are off.
And yes, it is a headless setup.

thrusty66, 2018-09-27:
Excuse me, everyone, but I made a very stupid mistake. 
Now the RaspBee works.
So I put RaspBee on the RPi a few days ago, I didn't have a light source yet. My first login to the gateway was supposed to be a Pir. As I now know, this can only be done by a small detour without a light source. 
At that time the red LED on the RaspBee was still lit. Then I read and tried a few things before I got in touch here. 
Anyway, I read this text incompletely: 
```
$ sudo raspi-config

() Interfacting Options > Serial

    * Would you like a login shell accessible over serial?
      No
    * Would you like the serial port hardware to be enabled?
      Yes
```

I answered the first question with yes and thus switched off the RaspBee. Therefore I couldn't get a connection anymore.
So my mistake.
But many thanks for the help and the patience.

closed 2018-09-27 by thrusty66

manup, 2018-09-27:
Hehe no worries, good that it works now 🙂

Pedder007, 2020-02-29:
Hi, I've got a propper running Raspi 3B with a deConz II since roughly a year. It connects my Pimatic HA with the ZigBee world (Hue bulbs, a couple of Xiaomi sensors and also some Tradfri devices). All works more or less propper and stable. Up to know I was always able to bind the devices by the Phoscon App and discover them afterwords in my HA.
As a further enhancement, I currently bought a light- and motion/light-sensors (Xiaomi GZCGQ01LM and Phil Hui motion indoor), but the setup of both now drives me a bit crazy since some days.
What I alredy found is, that there are still existing threads (dresden-elektronik #1559  and #2380 ) which show, that they are both tricky to connect and that some parameters have to be configured manually via the deConz Gui (not possible via the Phoscon App!).

**Now my question**, as I'm running Deconz in headless mode on the Raspi:
Is it possible to access the deConz GUI, e. g. from a PC having installed the deConz App on it, without changing anything in my Raspi/deConz configuration?
As I'm not that deap in the technics, I fear a little bit to crash my HA setup by switching the mode of deConz from headless to 'incl. GUI' mode. I found another thread (#2045 ) with a similiar question, but it seems not have been answered in terms of a 'simple' solution, which the usage of docker definitely would not be for me 

Happy for every hint to tackle this 😄 
Hopeful regards Pedder

CapitalT, 2020-09-14:
@Pedder007 thanks for bringing this thread on topic again. I would be interested too.

superjaegermaster, 2021-02-01:
Hello,

Is there no way from a PC or a Mac that has deconZ app install to connect remotely and fetch the mesh map ? X11 didnt work for me in stretch or Im really not using it correctly.

I think the quick win would be being able to conenct from the windows/mac app to a remote GW and exploit the mesh map from there

I would like to leave my RPI without GUI and keep it low (and please dont tell me I can install & remove GUI :) )

thanks in advance for your support

superjaegermaster, 2021-02-01:
nvm, I posted a feature request hoping it will be taken in account :)

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4326