dresden-elektronik/deconz-rest-plugin #801: 2.05.39 & Firmware 26240500

olemr, 2018-09-23:
Just updated  successfully on my Ubuntu Server Box.
Only thing found so far is that the date displayed in PWA is stuck in august :-)
But firmware version is now showing. Nice!

[image]

Is this firmware believed to fix Ikea lights becoming  unreachable after a few days?

manup, 2018-09-23:
The August issue is in the ever lasting browser cache ;)

You can force an update by appending `?abc123` in the browser URL bar once (where abc123 is just a random string).

I don't think the version will help for the IKEA issue here, have you checked which versions the bulbs have, maybe they can be updated?

olemr, 2018-09-23:
> You can force an update by appending ?abc123 in the browser URL bar once (where abc123 is just a random string).
That did the trick :-)

My lights on deconz report:
"modelid": "TRADFRI bulb E27 WS opal 980lm",    "swversion": "1.2.217",
"modelid": "TRADFRI bulb GU10 WS 400lm",    "swversion": "1.2.217",
"modelid": "TRADFRI bulb E27 CWS opal 600lm",    "swversion": "1.3.002",

Similar on Ikea Gateway:
White spectrum bulbs 1.2.217 (E27 & GU10)
Color/White spectrum bulbs 1.3.002

So, yes, they are all up to date. (Checked all)
Guess it is sniffing time. Have to buy another Conbee though. Will take a week or so.
Any hints of what to look for?
First sign of trouble is always after 48hrs, so I'm thinking maybe some kine of wraparound issue?
Have a total of 33 Ikea lights online. 16 of them are scattered pairwise around the facade. 12 of them are group fairly close together in the middle of the house and the rest is in different locations.
Lately it is always a facade light that fails first, usually followed by the Color bulb that is only 2m from the Conbee. Only the facade light is truly unreachable. The others are flagged as unreachable but still respond to group commands.
When bulbs are in this state, restarting deconz has no effect. Bulbs must be power cycled.

olemr, 2018-09-25:
Another Conbee on its way in the mail.
Will set up a sniffer.
I'm thinking maybe sniffing out the Ikea net first for comparison.
Any other sniffing hints as to what to look for is most welcome.

manup, 2018-09-26:
The most important part is to capture what happens when a working light turns unresponsive. Basically all commands going in and out the device. Also route requests targeting the lights are of interest.

closed 2019-01-24 by manup