dresden-elektronik/deconz-rest-plugin #802: > 48 hours unavailability

andreibanaru, 2018-09-24:
My rpi3 gateway was offline for more than 48 hours due to an SD card failure.
I've reinstalled everything on a new SD card and restored settings by using the Phoscon load backup option.

All my sensors and switches are listed as unavailable and no lights are shown in the WUI.

I've looked inside the zll.db and all entities are there.

What can I do besides repairing 19 lights, 5 switches and 7 sensors?

[image]

Also, for some reason Phoscon WUI and the old WUI list the gateway as using the channel 0?!

[image]

[image]

manup, 2018-09-24:
It looks very much like there is no connection between RaspBee and deCONZ.
Maybe the the serial port isn't configured yet?

Please check these steps:
https://github.com/dresden-elektronik/deconz-rest-plugin#software-requirements

andreibanaru, 2018-09-24:
You are so right. I've moved to hassOS and forgot to migrate the config.txt settings.

After adding `dtoverlay=pi3-miniuart-bt` into config.txt and rebooting RPI all entities are displayed in Phoscon.

Thank you!

closed 2018-09-24 by andreibanaru