dresden-elektronik/deconz-rest-plugin #803: Xiaomi aqara one button switch not showing in phoscon or webapp

Jasonthefirst, 2018-09-24:
I also wrote in another old thread about this but I think this warrants a new issue as well. I am running V2_05_39 and all my lamps and sensors are working (great work especially with the aqara cube. This is amazing). But I have three aqara single button switches (WXKG11LM) with a gyroscope. They are visible in the deconz gui and they seem to send something. But I can't find them in the phoscon app or the old webapp. Is there anything I can do? Or would some logs or something like this work? They are not visible in a REST get request as well. 
Thank you and if you need any more infos let me know.

Jasonthefirst, 2018-09-24:
The REST-api output doesn't include the switch as far as I can tell. There are only the sensors I can also see in the phosbon app.

manup, 2018-09-24:
Hmm I thought it was already in the REST-API. Can you please provide the screenshot of the *Basic Cluster* attributes like described in [Request Device Support](../wiki/Request-Device-Support)

You may need to repair the switch or press the setup button so the *modelid* will be send to the gateway.

Jasonthefirst, 2018-09-24:
Will do in a few hours as soon as I am on site. Can't access it remotely.

Jasonthefirst, 2018-09-24:
So, Here are the device informations. I hope this helps. And thank you for your time. 
[image]
[image]

manup, 2018-09-25:
Hmm are you sure about the WXKG11LM? From the issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/448 I read that the new switches with gyro are WXKG12LM.

The modelid `lumi.remote` isn't currently listed in the plugin I'm unsure if it is the same as the `lumi.sensor_switch.aq3`.

Jasonthefirst, 2018-09-25:
I am pretty sure as this is written on the back of the sensor. Should I send a link or photo?

manup, 2018-09-25:
>I am pretty sure as this is written on the back of the sensor. Should I send a link or photo?

Interesting, I'll check some other posts and projects if there are some hints about multiple model identifiers, we have seen other Xiaomi devices which reported different model identifies depending on wherever the value was reported on join or read by command.

manup, 2018-09-26:
I've added some code to address the model id variants https://github.com/dresden-elektronik/deconz-rest-plugin/commit/63bc05a883220f3a19fd415901f1510bdcda30ba

My switch is also labeled WXKG11LM but lacks the long press and shake parts and has model id lumi.sensor_switch.aq2.

Jasonthefirst, 2018-09-27:
Thank you for your support. 
I compiled your commit like it is described in https://github.com/dresden-elektronik/deconz-rest-plugin, copied it and tried to pair the remotes. 
I still can't see the remotes but I get this message in the console: 
[image]

manup, 2018-09-30:
Small update on the issue:
The switch is now on the REST-API and can be used in rules. Support in the Phoscon App and switch editor needs a bit more refactoring when I thought, I'm working on it but it will take a few more days until finished. In the future adding more of these very custom devices should be easier after refactoring is done.

vandalon, 2018-09-30:
should this work with the latest version or just only when compiling the master branch?

manup, 2018-09-30:
Only by compiling the master branch, or wait for version 2.05.40.

vandalon, 2018-10-01:
I might just wait then, this probably won't be long anyway :)

vandalon, 2018-10-07:
So, I have not waited and compiled :) It does work but I only see 2 and 4 response if i'm not mistaking I should get 12347 right?

vandalon, 2018-10-07:
I think I figured it out. so far deconz seems to assume that lumi.sensor_switch.aq3 and lumi.remote.b1acn01 are equal,  but this seems not to be the case. I got this sensor working with as '1234' with the following patch (Sorry, i'm not good with git and githun to quickly create a pull request , hope this helps though)

```
--- de_web_plugin.cpp.orig   2018-10-07 21:44:08.366853507 +0200
+++ de_web_plugin.cpp       2018-10-07 21:44:00.798542914 +0200
@@ -5285,7 +5285,7 @@
                                     else if (rawValue & 0x0100)  { buttonevent = side * 1000; }                // push
                                     else if (rawValue & 0x0200)  { buttonevent = side * 1000 + side; }         // double tap
                                 }
-                                else if (i->modelId() == QLatin1String("lumi.sensor_switch.aq3") || i->modelId() == QLatin1String("lumi.remote.b1acn01"))
+                                else if (i->modelId() == QLatin1String("lumi.sensor_switch.aq3"))
                                 {
                                     switch (rawValue)
                                     {
@@ -5297,6 +5297,17 @@
                                         default: break;
                                     }
                                 }
+                                else if (i->modelId() == QLatin1String("lumi.remote.b1acn01"))
+                                {
+                                    switch (rawValue)
+                                    {
+                                        case   1: buttonevent = S_BUTTON_1 + S_BUTTON_ACTION_SHORT_RELEASED; break;
+                                        case   2: buttonevent = S_BUTTON_1 + S_BUTTON_ACTION_DOUBLE_PRESS;   break;
+                                        case   0: buttonevent = S_BUTTON_1 + S_BUTTON_ACTION_HOLD;           break;
+                                        case 255: buttonevent = S_BUTTON_1 + S_BUTTON_ACTION_LONG_RELEASED;  break;
+                                        default: break;
+                                    }
+                                }
                                 else if (i->modelId().startsWith(QLatin1String("lumi.ctrl_ln")))
                                 {
                                     switch (event.endpoint())
```

Jasonthefirst, 2018-10-11:
Hi, I think I should give an update because I am using the beta 2.05.40 since it was online. Sadly I still can't see the switch in the phoscon app. 
I can see the buttonevent ("1002") in the rest api but it won't show phoscon. 
It gets a little bit weird name ("lumi.remote.blacn01 34") which makes it difficult to read it in homeassistant because it won't find the events from this device - but this is probably another problem.

Jasonthefirst, 2018-10-15:
@cnrd ahhh thank you. I tried the underscore at the end but didn't know the dots would be removed. Now I can finally use the switches as intended. Thank you all very much. 
They are still not showing in the phoscon app though.

closed 2018-10-15 by Jasonthefirst