dresden-elektronik/deconz-rest-plugin #807: IKEA Trådfri motion sensor battery

ebaauw, 2018-09-26:
Recently, my IKEA Trådfri motion sensor started to report motion at random moments, when there's none.

The sensor's `config.battery` reports 47 (%).  Both batteries measure at 3.02V.  Measuring the delivered current shows ~30mA for each battery, with the meter set to 1.5V (it only has 1.5 and 9V settings).  For a new battery, it's about ~64mA.  I replaced the batteries with new ones, to see if that solves the erroneous  motion detections.

What's you experience with these sensors?  At what battery level to they begin to act strangely?  We assumed the sensor reports battery percentage (0x0001/0x0021) in whole percents instead of half percents (as specified in the ZCL spec), but I think the batteries are way emptier than 47%.  I wonder what the Trådfri gateway reports.

manup, 2018-09-26:
>What's you experience with these sensors? At what battery level to they begin to act strangely?

The very important point with all PIR motions sensors is that they need to be positioned carefully, otherwise they could trigger on their own, for example when near a heater or any kind of cold/warm air circulation.

I had some really spooky moments in the past because of that :)

Here are some illustrated hints: https://support.somfyprotect.com/hc/en-us/articles/115005289985-Positioning-and-installing-the-motion-sensor

>We assumed the sensor reports battery percentage (0x0001/0x0021) in whole percents instead of half percents (as specified in the ZCL spec), but I think the batteries are way emptier than 47%.

I had to replace the battery of my IKEA sensor once (triggered like 6 times per day). 

> I wonder what the Trådfri gateway reports.

For these devices the only reliable view would be showing something like "battery ok" or "battery empty soon", percentage is very tricky.

ebaauw, 2018-09-26:
> The very important point with all PIR motions sensors is that they need to be positioned carefully

Yes.  The sensor had been in the same position for months, before the false motion detections appeared.  At first, I did consider the neighbours across the street playing with a laser pointer, but the detections still appeated with the blinds closed.  I locked the sensor in a dark cupboard to make sure it couldn't detect any real motion, and they still appeared.  Then I figured it would probably be battery related.

> I had to replace the battery of my IKEA sensor once (triggered like 6 times per day).

I wasn't quite there yet, but it seemed the rate was increasing from twice a week to a few times per day.

closed 2018-09-26 by ebaauw