dresden-elektronik/deconz-rest-plugin #808: Xiaomi Cube Rotation

Fl0ca, 2018-09-26:
Hey everyone, 
a friend is using deconz since yesterday, now there is a Problem with the Cube, which i dont own by myself. 

He cannot get any values from rotating the Cube, like a Poti. 
He uses 2.05.39 with 26240500 on the Conbee. 

Are there any known issues?

Oh-Noo, 2018-09-27:
I do not see the rotation in the Phoscon app

greet

matthias-schulz, 2018-11-02:
> 
> 
> I do not see the rotation in the Phoscon app
> 
> greet

Same here. The Rest-API does not report any rotation.
deconz Version: 2.05.44
RaspBee Firmware: 262D0500

Smanar, 2018-11-11:
I have same problem, take a look on uniqueid, it's a combinaison of mac adress + endpoint + cluster
For a working cube you need
00:15:8d:00:02:7c:f6:bf-02-0012
00:15:8d:00:02:7c:f6:bf-03-000C

and ATM I have
00:15:8d:00:02:7c:f6:bf-02-0012
00:15:8d:00:02:7c:f6:bf-01-0012

So I think sometime the API don't take the good endpoint in consideration.

matthias-schulz, 2018-11-14:
Okay, I solved it for myself. I had to connect the cube several times via Phoscon and it worked afterwards.

closed 2019-03-21 by stale[bot]