dresden-elektronik/deconz-rest-plugin #810: Raspbee failing on large payloads

rapture216, 2018-09-27:
I'm using a Raspbee, deCONZ with a Qt plugin, and a python script using the deCONZ serial protocol to play around with zigbee devices.  I noticed a weird behavior on deCONZ and confirmed it with my python: the Raspbee doesn't want to send messages with a payload above a certain size, even though the spec says it should. I double checked that I had the right firmware version (0x26210500), but this gets a Success status from the APS:
```12	cmd
13	seq
00	reserved
6800	Frame Length
6100	Payload length
13	seq
00	flags
02	dest mode
dec0	dest
00	dest ep
adde	profile
efbe	cluster
01	src ep
5200	payload len
45454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545
00	tx options
00	hops```

But this gets an Invalid Value:
```12	cmd
0f	seq
00	reserved
6900	Frame len
6200	Payload len
0f	seq
00	flags
02	dest mode
dec0	dest
00	dest ep
adde	profile
efbe	cluster
01	src ep
5300	payload len
4545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545454545
00	tx options
00	hops```

Is there something I'm missing?

rapture216, 2018-09-28:
@manup the deCONZ Serial Protocol says that the ADSU length should be 0-127, but it looks like it gives an INVALID_VALUE response from the APS for any ADSU longer than 82.

manup, 2018-09-28:
82 seems legit, the documentation must be clarified. 127 is the maximum total length of a transferred 802.15.4 frame. Where the first (127 - 82) bytes are used by MAC, NWK and APS layers for headers and encryption.

rapture216, 2018-10-15:
This is the longest message I've successfully sent from one raspbee and successfully received on another raspbee:

```
12    cmd
0a    seq
00    flags
5a00  frame len
5300  payload len
09    seq
00    res
02    mode
dec0  dest
00    dest ep
adde  profile
efbe  cluster
01    src ep
4400  data len
74686520717569636b2062726f776e20666f78206a756d7073206f76657220746865
206c617a7920646f673b20737068796e78206f6620626c61636b2071756172747a20
00    tx options
0a    hops
```

It ends up being a frame of 90 bytes instead of 127, and data of only 68 bytes (instead of 82). I found this because I was trying to verify the "fragmentation allowed" flag (0x08 in tx options) and it looks like the sending APS returns success but the message isn't received on the remote end. Something is going on between handing large messages to the APS and transmit/receive on the raspbees.

manup, 2018-10-16:
Fragmentation is not supported by the firmware (the stack has the option but it never was stable).

127 bytes is on the 802.15.4 PHY layer, have you checked in a sniffer to see the full frame?

rapture216, 2018-10-19:
So, I'm using a Killerbee to sniff my Zigbee and putting that through wireshark. When I send the biggest packet I can through the Raspbee this is what gets sent (edited to remove encryption):

```
6188a5aac2dec0cdbe4802dec0cdbe0af82879090000b37801ffff2e2100000000efbeadde01f374686520717569636b2062726f776e20666f78206a756d7073206f76657220746865206c617a7920646f673b20737068796e78206f6620626c61636b2071756172747a202b5df4edf86d
```

Wireshark breaks it out a bit:

```
6188    IEEE frame control 
a5      sequence number
aac2    PAN ID
dec0    destination
cdbe    source
4802    zigbee frame control
dec0    destination
cdbe    source
0a      radius/TTL
f8      sequence number
28    security flags
79090000  frame counter
b37801ffff2e2100 extended src
00      key index

00      frame flags
00      dst endpoint
efbe    cluster
adde    profile
01      src endpoint
f3      counter

(68 bytes of data)
74686520717569636b2062726f776e20666f78206a756d7073206f76657220746865206c617a7920646f673b20737068796e78206f6620626c61636b2071756172747a20

2b5df4ed  MIC
f86d    FCS
```

The whole thing is 113 bytes. I'm guessing the "missing" bytes are to account for using MAC addressing instead of NWK addressing, but (1) I haven't been able to get the raspbee to send anything except 113 bytes even with MAC addressed requested, and (2) if we used 32 bit addresses instead of 16 bit addresses we would only need, what, 8 more bytes not 14?

closed 2019-02-23 by stale[bot]