dresden-elektronik/deconz-rest-plugin #811: Cant get WXKG02LM OR WXKG03LM to pair

fillwe, 2018-09-27:
Been trying for 8 hours now to get these into my Conbee. Have tried following the advice on how to reset and pair from other threads here but with no luck. Have also tried to reset them via Touchlink but they are not showing up there either. Have also tried changing batteries. Running 2.05.39 via Hass.io on a Raspberry Pi, any ideas?

fillwe, 2018-09-27:
I'm starting to believe that there has been a new version of this realesed in August but with the same product name and . All listings on Gearbest, Aliexpress and Banggood are getting renewed and the older listings sold out. The packaging i got seems different from the old ones. https://www.banggood.com/Original-Xiaomi-Aqara-Wireless-Smart-Remote-Control-Switch-International-Version-Smart-Home-p-1341453.html?rmmds=search&ID=521034&cur_warehouse=CN

ErnstEeldert, 2018-11-01:
Looks like a different model identifier than described [here](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi-WXKG02LM). I just received these switches today, so assuming it's the new model.

[image]
[image]
[image]

fillwe, 2018-11-09:
Does not seem like we are getting any response, @ErnstEeldert could you post this in a new thread with device integration tag? We can close this one

manup, 2018-11-09:
The issue is fine, to track the development of WXKG02LM OR WXKG03LM. It just takes time due all the other ongoing topics.

manup, 2018-11-09:
Added pages to separate the WXKG02LM 2016 and 2018 versions.

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi-WXKG02LM-2016

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi-WXKG02LM-2018

Can anybody please provide logs for WXKG02LM 2018 version when pressing buttons?

    deCONZ --dbg-aps=2 --dbg-zcl=1 --dbg-info=1

goermezer, 2018-11-09:
Here is an output of:

- Press left button
- Press right button
- Press both buttons
- Double press left button
- Double press right button

Sensor which reports:  0x00158D00027C277E

Attached file:

[0x00158D00027C277E.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2566587/0x00158D00027C277E.txt)

Hope, it helps. I have 43 Nodes, which make a lot of noise...

o0shojo0o, 2018-11-10:
I tried to match goermezer: 

- Press left button
- Press right button
- Press both buttons
- Double press left button
- Double press right button

Sensor which reports: 0x84182600000c787d

[0x84182600000c787d.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2568352/0x84182600000c787d.txt)

ebaauw, 2018-11-10:
> Sensor which reports: 0x84182600000c787d

That’s an OSRAM device, not a Xioami.

goermezer, 2018-11-10:
Thanks @ebaauw, you are right. It wasn`t the right id. I will test again...

goermezer, 2018-11-10:
Ok, here is another test. The reporting sensor is 0x00158D00027C277E20.
[0x00158D00027C277E20.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2568664/0x00158D00027C277E20.zip)

o0shojo0o, 2018-11-11:
> > Sensor which reports: 0x84182600000c787d
> 
> That’s an OSRAM device, not a Xioami.

Oh sorry, I caught the wrong device.
Am still quite new in the matter, and therefore have problems in the log to find the right entries.

samsonovss, 2018-11-15:
Version - 2.05.47 WXKG02LM not work
[image]
[image]

manup, 2018-11-15:
They are not party of .47, I'm expecting the switches and some other Xiaomi to arrive next week and hopefully to complete, test and support for them in the API and Phoscon App quickly.

goermezer, 2018-11-18:
@manup I saw .48 today, installed the version and tested again my WXKG02LM 2018. But the device is still not detected in REST API. With deconz GUI it joins the network and is identified as lumi.remote.b286acn01.

o0shojo0o, 2018-11-27:
Hi manup,
have you already received your Xiaomi delivery?
And if so, is there something new?
Here are a few WXKG02LM waiting for their use ;)

manup, 2018-12-02:
The switches are still not here :/ but I've started work on it based on the data provided here.

carloscae, 2018-12-03:
@manup the unsung hero. thanks a lot.
do you plan on supporting the double and long presses as well? even if not initial release?

manup, 2018-12-03:
If the switches support these yes, the above logs didn't indicate that. If you can provide this data the related events can be added quickly.

carloscae, 2018-12-03:
The new switches do support single, double and long press. They are accessible via Aqara official hub.

I unfortunately have no windows machine to try the Deconz interface, but I'll get a virtual machine up and running to get those logs. I have both single and double rocker remotes.

greencharlie100, 2018-12-04:
I am running deconz 2.05.49 on hassio. Is this the version with the updates to support the WXKG03LM wireless switches?
If so - how do I pair the switch with Phoscon?

carloscae, 2018-12-04:
No. It isn't. Probably on 2.05.50

carloscae, 2018-12-04:
@manup only send the commit for the update, but the new version was not yet released. once it is, we need to ask or wait for marthoc to update his Hassio plugin. Might be necessary to also update the firmware of your Conbee/Raspbee.

greencharlie100, 2018-12-04:
Sorry, I did not realise that. I will wait patiently for the Hassio plugin update. Thanks for the info.

carloscae, 2018-12-04:
I'm also looking forward very much to this. I currently have it paired directly on Homekit and its automations are extremely crappy. Anyway, you can follow the Hassio plugin development here:

https://github.com/marthoc/hassio-addons

greencharlie100, 2018-12-04:
I will keep an eye on it. Thanks again.

manup, 2018-12-04:
[copy and paste]
We are working on the Phoscon App Switch editor support for the following switches:

* WXKG01LM
* WXKG11LM
* WXKG12LM
* WXKG02LM
* WXKG03LM

Some of these have different versions wich will also be handled by the Switch Editor.

Current plan is to get a first preview online till Saturday, since I still don't have all switches here for testing there might be issues which need to be ironed out in further releases.

Phoenix-DH, 2018-12-05:
Hey guys,

I got a ConbeeStick running since 3 days. Everything is working fine, except this xiaomi switch WXKG02LM.

I read a lot in this fork but am not sure if it normally should work or not.
On my package there was only something mentioned with a timestamp from 2016 ... I read nowhere something from 2018.
Should my version work?

I tried connecting as switch, as sensor, no response inside PHOSCON.
I tried to connect via reset of the switch, pushing both for 10 seconds until the LEDs will blink ...
Or am I doing something wrong?

I installed the deb from http://www.dresden-elektronik.de/rpi/deconz/deconz-latest.deb which should be the same than this one here, right? Is version 2.05.49.

Thanks!

carloscae, 2018-12-05:
One way to know if you have a newer model is by the looks of the box. Does it have only a gray Aqara logo or is it a nicer box with a photo of the product?

carloscae, 2018-12-05:
You have the new one which is not yet supported, so your issues as expected. Good news is that it should be fixed on the next release some time this week. I’m also on this boat. Now we have to wait just a little bit more. :)

Phoenix-DH, 2018-12-05:
Okay, then my assumption is correct.
Thank you!

How can I then update my version from:
http://www.dresden-elektronik.de/rpi/deconz/deconz-latest.deb

Will this work directly via PHOSCON App like the initial FW Update?

carloscae, 2018-12-05:
I assume once your case you can do that directly via Phoscon app once it's officially released: Keep an eye for 2.05.50. 
I only have experience with Hassio installation which is a bit more complicated to update than normal installations.

OmikronApex, 2018-12-05:
@manup I might not have logs for the new WXKG02LM Switches yet, but I got my deconz hooked up to iobroker and the switch reports the following button events:
x001
x002
x004
for left, right and both buttons respectively as 1001, 2001, 3001 etc. 
No x003 sadly as it seems.

Do you still need exact deconz logs to add the events? I might be able to set up a clean version of deconz to get them later this day.

manup, 2018-12-05:
>Do you still need exact deconz logs to add the events? I might be able to set up a clean version of deconz to get them later this day.

Yes, the output with what is received would is needed.

    deCONZ --dbg-info=1 --dbg-aps=2 --dbg-zcl=2

OmikronApex, 2018-12-05:
@manup Alright, I'll post the logs once I got them.

OmikronApex, 2018-12-05:
Didn't hav the time to do a clean setup of deCONZ but here are my logs for the new WXKG02LM:

- left single
- left double
- left hold
- right single
- right double
- right hold
- both single
- both double
- both hold

The corresponding node is 00158D00027C1631.

[00158D00027C1631.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/2650349/00158D00027C1631.txt)

Hope this helps :)

manup, 2018-12-05:
Thanks, I'm afraid the payload is missing was the log done with following debug options?

(note level 2)

    deCONZ --dbg-info=1 --dbg-aps=2 --dbg-zcl=2

OmikronApex, 2018-12-05:
Yes I used these exact parameters

manup, 2018-12-05:
First line in the log says `pi@phoscon:~ $ deCONZ --dbg-info=1 --dbg-aps=1 --dbg-zcl=1 --dbg-zdp=1 --http-port=80`

The needed information is only visible with ` --dbg-aps=2`

manup, 2018-12-06:
Thanks this one works and also my switches and new sensors arrived today.

Just took a look at the WXKG02LM (lumi.remote.b286acn01) in the sniffer and it's weird ;)

Following happens for left, right and both buttons:

* Single click: multistate value 0x0001
* Double click: multistate value 0x0002
* Hold: multistate value 0x0000,
* ... followed by an report for Basic Cluster on attribute 0xfff0 with 9 character octed string  `aa:10:05:41:87:19:01:10:01`
* **Important** No Hold release indication is sent!
* Hold for more than 10 seconds the switch **leaves** the network and is no longer available

Therefore it won't be possible to support push dim like:
* hold: dim lights 
* release stop dim

I think it's dangerous to use the hold event since the switch does a reset by just pressing any button for greater than 10 seconds, but the same is true for the old WXKG02LM version.

manup, 2018-12-09:
Version 2.05.50 supports more switches in the REST-API, Phoscon App switches view and the switch editor.
https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_50

Support for double click events for the wall switches will be added in the next version.

Phoenix-DH, 2018-12-09:
Thanks for the great work.
How can I perform the update?

I had a llok to the phoscon gsteway page. There is after esch reboot the message rhat there is a new firmware: 262F0500 although this version is alresdy installed.

But no hint about new version.

manup, 2018-12-09:
Currently you need to manually update the version, update via app will be shown in a few days.

https://github.com/dresden-elektronik/deconz-rest-plugin#precompiled-deconz-packages

Phoenix-DH, 2018-12-09:
Okay, thank you!

Can I follow this steps, only for the package?

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

For sure, take the latest package.

manup, 2018-12-09:
Should be enough when you do steps 1–3 mentioned from here

https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz

Phoenix-DH, 2018-12-09:
Worked, thanks.
Had to restart the service to update version in GUI.

koen01, 2018-12-09:
4x WXKG02LM 2018 version tested and working. 
Added via phoscon. 

Thank you!

Phoenix-DH, 2018-12-10:
I also get mine paired as switch.
In homekit with homebridge-hue plugin the switch is not recognized as switch.

Is this working for anybody? Some hints?

manup, 2018-12-10:
>In the meanwhile I added a light to PHOSCON and the switch was not working too...

Can you please provide more information?

Phoenix-DH, 2018-12-10:
What do you exactly need?

This are steps I did for PHOSCON:
I deleted one lamp in hue bridge, added via PHOSCON.
Created a group with this one lamp and the switch.
Added an action to each of the both buttons.
Pushed the button, but nothing is happening.

Phoenix-DH, 2018-12-10:
I reconfigured everything, now within the phoscon app everything is fine.
Everything was also exposed to homekit.

Only after a period of time now I got the following:
```
[2018-12-10 16:47:05] [ConbeeStick] ConbeeStick: gateway error 3: resource, /lights/3, not available
[2018-12-10 16:47:05] [ConbeeStick] Schlafzimmer Fluter: homekit on changed from 0 to 1
```

Will have an eye on it ...

Switch the light on/off via hardware switch works as well as per PHOSCON ...
Only homekit is stucking now.

manup, 2018-12-10:
The json looks fine, just testet websocket events are also fired.
@Kane610 Maybe the `modelid` isn't yet in supported in Home Assistent component?

Kane610, 2018-12-10:
If the battery shows up as a sensor it works as expected since remotes are only exposed as events in home assistant.

carloscae, 2018-12-10:
> I seem to have the same issue, with the switch being recognised in Phoscon, but somehow Home Assistant doesn't see the switch. Just the battery status.
> 
> If I do a get on /api/[key]/sensors, the switch shows up. So I don't know whether it's Home Assistant that doesn't parse the json correctly, or Phoscon/deConz that replies with wrong json data.
> 

Light switches will only trigger events.
You need to use automations like the example bellow:

`- alias: Aqara Toggle
  initial_state: 'on'
  trigger:
    platform: event
    event_type: deconz_event
    event_data:
      id: aqara_1
      event: 1002
  action:
      data_template:
        entity_id: light.office
      service_template: >
        {% if is_state('light.office','off') %}
          homeassistant.turn_on
        {% else %}
          homeassistant.turn_off
        {% endif %}`

greencharlie100, 2018-12-10:
Is the latest update available in hass.io? Or  is there any way I can implement this latest update myself in hass.io?

greencharlie100, 2018-12-10:
Apologies - I see the latest deconz hass.io addon has been released 25mins ago...

Phoenix-DH, 2018-12-11:
> I seem to have the same issue, with the switch being recognised in Phoscon, but somehow Home Assistant doesn't see the switch. Just the battery status.
> 
> If I do a get on /api/[key]/sensors, the switch shows up. So I don't know whether it's Home Assistant that doesn't parse the json correctly, or Phoscon/deConz that replies with wrong json data.
> 
> `"2": { "config": { "battery": 100, "on": true, "reachable": true, "temperature": 2300 }, "ep": 1, "etag": "2c397e05aa2fecab16740c872cbcdabd", "manufacturername": "LUMI", "mode": 1, "modelid": "lumi.remote.b286acn01", "name": "Schakelaars Boven", "state": { "buttonevent": 1002, "lastupdated": "2018-12-10T20:19:23" }, "type": "ZHASwitch", "uniqueid": "00:15:8d:00:02:7c:0d:e2-01-0012" },`

I did everything from the scratch...
Now it seems to work. I have no outages in Homekit until now.

I use the homebridge-hue plugin to expose the switches.
At the moment they are only known as unknown component, not shown like the original hue switch.
I can image that that will also came up within the next days.

Phoenix-DH, 2018-12-11:
I do not have those values ... maybe somebody can provide them?
I already opened an issue for that yesterday:
https://github.com/ebaauw/homebridge-hue/issues/407

Maybe those values can be added.

Thank you all for the great work.

greencharlie100, 2018-12-11:
I am looking to do an automation – toggle a wired switch with a wireless switch event…

Here are details of both my switches:

Wireless Switch - *Sensor:
"2": {
            "config": {
                "battery": 100,
                "on": true,
                "reachable": true,
                "temperature": 2300
            },
            "ep": 1,
            "etag": "340e11d662f566ff4f0f883ceaeeba02",
            "manufacturername": "LUMI",
            "mode": 1,
            "modelid": "lumi.remote.b186acn01",
            "name": "Hall Lower",
            "state": {
                "buttonevent": 1002,
                "lastupdated": "2018-12-11T17:59:59"
            },
            "type": "ZHASwitch",
            "uniqueid": "00:15:8d:00:02:7c:3e:0e-01-0012"
        },

Wired Switch -*Light:
"1": {
            "etag": "42da2d16d2c49503441d4ee524fbbb2e",
            "hascolor": false,
            "manufacturername": "Unknown",
            "modelid": null,
            "name": "Hall Lower",
            "state": {
                "alert": "none",
                "on": false,
                "reachable": true
            },
            "swversion": null,
            "type": "Smart plug",
            "uniqueid": "00:15:8d:00:02:a1:8a:76-02"
        },

I was trying something along these lines provided by @carloscae above:

automation:
  alias: Hall Lower - Toggle
  initial_state: 'on'
  trigger:
    platform: event
    event_type: deconz_event
    event_data:
    id: 1
    event: 1002
  action:
    data_template:
    entity_id: lights.1
    service_template: >
        {% if is_state('light.1','off') %}
        homeassistant.turn_on
        {% else %}
        homeassistant.turn_off
        {% endif %}

Can anyone help make this automation work?

carloscae, 2018-12-11:
@greencharlie100 try using “hall_lower” as the ID on the automation. That’s normally how Hassio parses the Deconz names.

greencharlie100, 2018-12-11:
I tried changing that Id - but still not working.

[38 more replies]

closed 2019-08-03 by stale[bot]