dresden-elektronik/deconz-rest-plugin #814: Support for Plugwise Circle?? Will it work with deconz?

svartpippi, 2018-09-29:
Hi,
I bought a set of Plugwise Circles, with one Circle+ and a Plugwise USB stick.
They use the Zigbee protocol but it seems that they use it in a special way.
I have tried to get the data with the deconz software and a conbee stick but nothing shows up, to be able to send in for support of new device.
What I have learned is that to be able to setup the Circle plugs in a network, one need a Circle+ which communicates with the stick and configure the Circle devices. 
Is it possible to get them to work with deconz?

frankhommers, 2018-12-01:
@svartpippi, any progress on this?

svartpippi, 2018-12-02:
@frankhommers No, I have given up on getting them in with deconz.
I'm trying to get plugwise stick working in docker with https://github.com/SevenW/Plugwise-2-py as it looks to be working in a very strange way that don't suit deconz.

svartpippi, 2018-12-08:
@frankhommers I changed my mind and instead has decided to use openhab2 with plugwise binding and MQTT over to my main home-assistant setup, and it's working really good and was really easy to setup.

closed 2019-04-14 by stale[bot]