dresden-elektronik/deconz-rest-plugin #819: Support for Samsung Smart Outlet (powerplug)

clementx2006, 2018-10-01:
requesting through a dedicated post for the ease of tracking (and closing down quickly).  

I have got the UK version but assuming very similar with the ones for other regions. 

This is the plug: https://www.samsung.com/uk/smartthings/sensors-plug-f-app-uk-v2/

Screenshots:
**Clusters of the node** 
[image]

**Basic cluster**
[image]

**Electrical Measurement cluster**
[image]

**On/off cluster**
[image]

Please let me know if anything else is required. 
reference also #432 where the discussion had started. 

Many thanks in advance!

ebaauw, 2018-10-01:
The _Power Multiplier_ and _Power Divisor_ values suggest _Active Power_ is 9.5W so _RMS Current_ is 38mA?

clementx2006, 2018-10-01:
Correct. There was only a Raspberry Pi on the circuit when the measurement was taken.

flinthamm, 2018-10-25:
Can you confirm what is (if anything) working and what isn't with Deconz/Phoscon? Are you able to at least reliably switch the socket on/off?

clementx2006, 2018-10-25:
Can switch. But would like to have power current monitoring. And not appear as a light...

clementx2006, 2018-11-19:
Hello everyone - any update on when the power monitoring will be supported on this smart plug (also not appear as a light in Home Assistant)?

closed 2019-03-26 by stale[bot]

stigvi, 2019-10-22:
Still nothing?

clementx2006, 2019-10-22:
Same question as it’s been over a year and Samsung just released a new UK version. The support for usage monitoring is really what’s missing (and could be useful in so many scenarios).

clementx2006, 2019-10-22:
And not sure how to reopen this...

blair287, 2020-02-09:
Yea i also want power reporting why can it not do this?

smartypantsuk, 2020-02-09:
The new 2019 Samsung plug outlet is now partially supported meaning on/off switching and the monitoring of load (watts) is working. The 2019 plug is not the same plug aforementioned in this plug. 

While the plug is working in Deconz on Linux, the Windows version has not been updated yet. 

Full post here.
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2121#issue-527681957

SwoopX, 2020-02-27:
Should work with the next release.

smartypantsuk, 2020-02-27:
> Should work with the next release.

I'm not sure why the Windows version hasn't been updated in months. Prior to this year, it has been updated multiple times every month thought 2019. There's been no update for nearly 4 months now.

SwoopX, 2020-03-02:
It's not just Windows, that goes for all of us.