dresden-elektronik/deconz-rest-plugin #820: ZHAPower state.current in mA instead of 0.1A ?

ebaauw, 2018-10-01:
The smart plugs we knew when adding support for ZHAPower sensors reported current with a resolution of 1A or 0.1A.  Consequently, `config.current` reports current in 0.1A.  We now see a number of plugs that provide a finer-grained resolution, up to 1mA.  I would like to change `state.current` to report the current in mA.

Alternatively, we could report `state.current` in A, using a `float` value.  While this makes more sense logically, it isn't currently supported by the web socket notifications (that still report `state.x` and `state.y` as `int` instead of `float`).

If we should change this, it would make sense to use `float` values for a number of other attributes as well:
- `state.power` - currently in W, but we've seen plenty of smart plugs that support a resolution of 0.1W;
- `state.consumption` - currently in Wh, but kWh makes more sense;
- `state.voltage` - currently in V but there's smart plugs with a resolution of 0.01V, even though they don't measure voltage this accurately;
- `state.temperature` - currently in 0.01°C, plain °C makes more sense.  It would introduce a deviation from the Hue API, though;
- `state.humidity` - currently in 0.01%, but plain % makes more sense.

clementx2006, 2018-10-08:
All making sense although question is what we need to do to move this forward? I assume no new powerplug integration could be implemented while this remains unresolved?

ebaauw, 2018-10-08:
> I assume no new powerplug integration could be implemented while this remains unresolved?

Not necessarily.  Only the ZHAPower resource might show a coarser-grained resolution for `state.current` than the plug supports.