dresden-elektronik/deconz-rest-plugin #822: aqara water leak sensor

lastessa, 2018-10-02:
sensor is shown in logs but can add it via web pwa:
19:57:56:093 Node 0x00158D0002335001 is known by 1 neighbors, last seen 3 s
19:57:56:573 Node 0x00158D0002332E68 is known by 1 neighbors, last seen 0 s
19:57:56:779 MAC Poll 0x02 0x7ABB
19:57:56:812 ZCL attribute report 0x00158D0002332E68 for cluster 0x0000, ep 0x01
19:57:56:813    payload: 050042156c756d692e73656e736f725f776c65616b2e61713101ff42220121c70b03281b0421a8010521970006240000000000082104020a210000641000
19:57:56:813 0x00158D0002332E68 skip Xiaomi attribute 0x0005
19:57:56:813 0x00158D0002332E68 extract Xiaomi special
19:57:56:813    01 battery 3015 (0x0BC7)
19:57:56:813    03 temperature 27 °C
19:57:56:813    04 unknown 424 (0x01A8)
19:57:56:813    05 RSSI dB (?) 151 (0x0097)
19:57:56:813    06 LQI (?) 0 (0x0000000000)
19:57:56:813    08 unknown 516 (0x0204)
19:57:56:813    0a unknown 0 (0x0000)
19:57:56:813    64 on/off 0
19:57:56:813 APS-DATA.indication from child 0x7ABB
19:57:56:813 Node data 0x00158d0002332e68 profileId: 0x0104, clusterId: 0x0000
19:57:56:814 APS-DATA.indication from child 0x7ABB

Alexxander0, 2018-10-02:
I cant pair my new aqara water leak sensor either. Iam not sure if thats limited by the amount of sensor i have (10 at the moment), because i read about that limitation. Or maybe got the same problem like you. How do you get the logs?

lastessa, 2018-10-02:
just stop deconz service and run it manually with:
` /usr/bin/deCONZ -platform minimal --http-port=80 --dbg-info=2`

manup, 2018-10-02:
Is the sensor shown in the REST-API? Note: It currently isn't visible in the Phoscon App.

The recent version 2.05.39 with firmware 0x26240500 supports up to 32 directly connected sensors (was 10 before) each router device like a lights extends the limit further.

Alexxander0, 2018-10-03:
I already have one xiaomi leak sensor in deconz (paired it a a while ago). This working sensor doenst show in frontend as you sad, but is shown in the rest api. The new sensor doenst appear. Maybe the changed something in new sensors?

`11:23:28:698 MAC Poll 0x02 0x3329
11:23:28:794 device announce 0x00158D000233F478 (0x3329) mac capabilities 0x80
11:23:28:794 set fast probe address to 0x00158D000233F478 (0x3329)
11:23:28:795 device announce 0x00158D000233F478 (0x3329) mac capabilities 0x80
11:23:28:795 set fast probe address to 0x00158D000233F478 (0x3329)
11:23:28:795 APS-DATA.indication from unknown node 0x00158D000233F478
11:23:28:800 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 2, node: 0x3329
11:23:28:800 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 1, node: 0x3329
11:23:28:800 new node - ext: 0x00158d000233f478, nwk: 0x3329
11:23:28:818 Node 0x000B57FFFE11CA42 is known by 0 neighbors, last seen 0 s
11:23:28:842 ZCL attribute report 0x00158D000233F478 for cluster 0x0000, ep 0x01
11:23:28:843    payload: 050042156c756d692e73656e736f725f776c65616b2e61713101002004
11:23:28:843 0x00158D000233F478 skip Xiaomi attribute 0x0005
11:23:28:843 APS-DATA.indication from child 0x3329
11:23:28:843 verify 0x00158d000233f478 is child node after 828684 s
11:23:28:845 ~Resource() /lights 0x7ed27980
11:23:28:845 DB save zll database items 0x00000800
11:23:28:845 DB sql exec UPDATE devices SET timestamp = strftime('%s','now'), nwk = 13097 WHERE mac = '00:15:8d:00:02:33:f4:78';INSERT INTO devices (mac,nwk,timestamp) SELECT '00:15:8d:00:02:33:f4:78', 13097, strftime('%s','now') WHERE (SELECT changes() = 0);
11:23:28:862 DB saved in 16 ms
11:23:28:873 don't close database yet, keep open for 900 seconds
11:23:28:873 Node data 0x00158d000233f478 profileId: 0x0104, clusterId: 0x0000
11:23:28:890 verify neighbor status: APP_SUCCESS (0x00)
11:23:29:298 Node 0x00158D00022381BB is known by 0 neighbors, last seen 0 s
11:23:29:642 ZCL attribute report 0x00158D000233F478 for cluster 0x0000, ep 0x01
11:23:29:642    payload: 01ff42220121d10b0328190421a80105211f0006240100000000082104020a210000641000
11:23:29:643 0x00158D000233F478 extract Xiaomi special
11:23:29:643    01 battery 3025 (0x0BD1)
11:23:29:643    03 temperature 25 °C
11:23:29:643    04 unknown 424 (0x01A8)
11:23:29:643    05 RSSI dB (?) 31 (0x001F)
11:23:29:643    06 LQI (?) 4294967296 (0x0100000000)
11:23:29:643    08 unknown 516 (0x0204)
11:23:29:643    0a unknown 0 (0x0000)
11:23:29:643    64 on/off 0
11:23:29:644 APS-DATA.indication from child 0x3329
11:23:29:644 Node data 0x00158d000233f478 profileId: 0x0104, clusterId: 0x0000
11:23:29:778 Node 0x00158D0001B707E2 is known by 0 neighbors, last seen 0 s
11:23:29:845 [1] get node descriptor for 0x00158d000233f478
11:23:30:107 MAC Poll 0x02 0x3329
11:23:30:258 Node 0x00158D000233F478 is known by 0 neighbors, last seen 0 s
11:23:30:315 MAC Poll 0x02 0x3329
11:23:30:315 Incr. ZDP retry count 2 on item 3
11:23:30:364 APS-DATA.indication from child 0x3329
11:23:30:364 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 3, node: 0x3329
11:23:30:377 don't close database yet, keep open for 900 seconds
11:23:30:469 [2] get active endpoints for 0x00158d000233f478
`
Sorry, think i cant format that any better. There might be some messages from my other devices

Elliott32224, 2018-12-03:
Same problem here. Xiaomi leak sensor cannot be added with Phoscon. Although it shows up in the REST API, the deConz scanning won't finish.

simzdk, 2018-12-04:
Same problem here.

manup, 2018-12-04:
The Phoscon App view of measured values will be added soon when my new sensors arrive, also sensor search for this should be fixed then.

manup, 2018-12-09:
Just testet with my new water sensor, joining failed.

Reasons seems to be that the sensor has a IAZ Zone cluster but doesn't list it in the simple descriptor.
Working on a fix ...

kami83, 2018-12-14:
Hi together,

yesterday i found a very easy and good way to connect the Xiaomi water leak sensor to the conbee.
How i did it:
1. Open the deconz-gui.
2. Remove the old leak sensor node (if added)
3. Remove the battery of the leak sensor
4. wait 10 sec and plugin the battery again 
5. Open the old Webapp and go to Settings->Open Network
6. Press and hold the leak sensor button until the blue light starts to flash. Leave the button and put the sensor next to the conbee stick. 
7. After this i can see the sensor with the Name "Water11" in the gui.

I hope this will help you.

Cu kami

Elliott32224, 2018-12-14:
Darn! I could not get this to work by the above instructions. Elliott

manup, 2018-12-14:
2.05.51 will arrive soon, it should work way better with this version.

Elliott32224, 2018-12-17:
I have been waiting for 2.05.51 to be posted and was happy to see it today. I installed it and it is definitely an improvement, but still needs tweaking. I had 2 leak sensors to install and could only install one successfully. The second one installed as in earlier versions - it had no name associated with it in the deConz GUI. The one that installed successfully has the name "lumi.sensor_wleak.aq1." Neither shows up in the Phoscon App. However, with this new version, the Phoscon app shows the smart round button switch  (WXKG01LM). It names it "lumi.remote.b1acn01." Elliott

manup, 2018-12-17:
> This time it showed up in the deConz GUI as "lumi.sensor_wleak.aq1(2)." It still did not show up in Phoscon. But, it works!

Looks like the cached version is too old, to refresh the cache please append a bogus search query to the url

    http://<gateway ip>/pwa/devices-sensors.html?239874

Elliott32224, 2018-12-17:
Bogus search query did not make any difference for me.

manup, 2018-12-17:
Hmm strange, when you open developer console in the browser (F12) is there any error shown on the sensor site?

Also, can you please provide the REST-API output of the sensor which isn't shown?

Elliott32224, 2018-12-17:
Okay, so this is interesting. I have been accessing Phoscon from a different computer on my home wireless network than the computer that has the Conbee plugged into it. When I opened Phoscon on the computer with the Conbee and deConz, the sensors are listed. As you first thought, it may be related to the cache. So, this is all good. Thank you. Elliott