dresden-elektronik/deconz-rest-plugin #823: Creating rules for Trust ZYCT-202

jnmarczi, 2018-10-03:
Just updated my raspBee to the latest version (2.05.39) and to the latest firmware...  
Added my Trust ZYCT-202 and it's now visible at the ReST api, which is great!!!

As far as i understood, the current selected group of the ZYCT-202 is available at the group property of sensor's config (values 145 to 149 for groups/lights 1-6, and value 150 for all groups/lights).

The problem IMHO is, that adding a rule condition only works with the state of a sensor. am I wrong about this? 

is it possible to add the group property to the state of the sensor? or will it be possible to address a config value in the future?

jee67, 2018-10-03:
How did you manage to pair the ZYTC-202? I’m trying for days now, but still no success. I have the latest deconz and firmware version too.

jnmarczi, 2018-10-03:
Searched for a new sensor at the phoscon app
Removed the battery of the zyct-202 and inserted it again. then pressed & holt the button the 2 bulbs until all leds blinked.

jnmarczi, 2018-10-03:
Damn, i closed this issue

jee67, 2018-10-04:
For a ZYCT-202: New sensor or new switch?

jnmarczi, 2018-10-04:
Think i used "new sensor", but i haven't got a success message from the phoscon app, just the lights of the remote blinked.

jnmarczi, 2018-10-10:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/261

closed 2018-10-10 by jnmarczi