dresden-elektronik/deconz-rest-plugin #825: Need advice for 2.4 ghz!!

MattL0, 2018-10-03:
HI, 

This is not related directly to any deconz gateway, but i am facing issues with 2.4 ghz.

I am renting a basement in a house. The whole house is getting good wifi signal ( 2.4 and 5 ghz) everywhere , and the router is installed in the basement.  
But,
In a particular room of the basement, it has big issue  in the 2.4 ghz band, and the coonbee ( on channel 15 too. With wifi i get constant disconnection when i am in this room. 
So... i have installed a osram plug in this room for testing... and it is almost always unreachable in this room ( in a outlet very near of the entrance door)

I suspect something in interfering here . I have bought a lot a ''router '' but this do not fix the issue. It is ''funny that I get very good signal on the second floor of the house ( which is very far from the router), and in that room ( wich is on the same floor..) i get poor signal...

I could somewhat fix the wifi issues with an access point pluged with a powerline.. 
But for zigbee, i do not know what to do. 

For now, i am using zwave device in that room and it works great.

Do you guys have any suggestion for this?

Thanks,
matt

MattL0, 2018-10-03:
I would like to use Keen home vent in that room. But, this device is not cheap and i live in Canada. So, if it does not work.. It would be difficult to get a ''full'' refund.

MattL0, 2018-10-04:
yes

MattL0, 2018-10-04:
ok thanks for the graph. So i Could try channel 11 Conbee

MattL0, 2018-10-04:
Can't control the plug with channel 11. is there a possibility to add another conbee ? This would be plugged into a raspberry pi with VirtualHere https://www.virtualhere.com/usb_client_software ?

manup, 2018-10-04:
> yes

When the WiFi gets unstable it's unlikely that ZigBee is the cause, normally the other way around may happen for example when you stream Musik/Video over WiFi. ZigBee doesn't send enough data to disturb WiFi.

You can check this by just turning of the ConBee, and test if WiFi gets better.

>Can't control the plug with channel 11. is there a possibility to add another conbee ? This would be plugged into a raspberry pi with VirtualHere https://www.virtualhere.com/usb_client_software ?

Multiple ConBee dongles is not recommend since only one can be the coordinato, but you can extend the mesh with every router device like ZigBee lights and plugs.

To be honest I'm a bit confused what exactly is bad, the Wifi connection or ZigBee mesh?

manup, 2018-10-04:
When you might need more repeaters (Wifi and ZigBee) to improve the signal to the room.
Note: all radio signals get weak when walls and doors contain metal or glass.

closed 2018-10-04 by MattL0