dresden-elektronik/deconz-rest-plugin #828: How do I add the Xiaomi Leak Sensor?

SebSemmi, 2018-10-04:
I can't add this sensor in the Phoscon App.

manup, 2018-10-04:
The leak sensor is currently only visible in the REST-API.
It can be added by using Phoscon App > Menu > Settings > Sensors > Add Sensor > Other

SebSemmi, 2018-10-04:
Is there a reliable to pair the sensor? I couldn’t find one

manup, 2018-10-04:
For most Xiaomi sensors, while sensor search is running, you need to press a button until a LED flashes.
Also please make sure to use the latest version of deCONZ.

Alexxander0, 2018-10-11:
Did you made any progress? And other ppl too https://github.com/dresden-elektronik/deconz-rest-plugin/issues/822#issue-366056364

SebSemmi, 2018-10-11:
Nope, can't get it to pair. I tried everything I could find

ebaauw, 2018-10-11:
Note that you need to squeeze the sensor quite hard for pressing the reset "button".  It took me some time to figure that out.  Even then, Xiaomi devices are notoriously difficult to pair.

Try the following:
1. Power down all your ZigBee routers (lights), and bring the sensor within range of the RaspBee/ConBee (to force a direct connection to the sensor instead of over the mesh network);
2. Search for new devices from the Phoscon app.  Make sure to allow for enough time (10 min);
3. Squeeze the sensor for 10 seconds, until the light starts to blink - this should reset the sensor;
4. Wait for the sensor to stop blinking and appear in the deCONZ GUI (check the _Node List_ panel for a new record - sometimes the node is hard to find because it's overlapping with an existing node);
5. Start briefly squeezing the sensor every 2 seconds or so (to keep it awake) - this should populate the clusters if not already present;
6. Open the _Cluster Info_ panel and select the sensor's _Basic_ cluster.  Read the attributes, while continuing to squeeze the sensor every two seconds or so;
7. When the attributes are read (notably the _Model Id_), the REST resource is created.

Webserve, 2018-10-14:
I have the same problem. @ebaauw your solution didn't work. Other sensors from aqara are working fine.

ebaauw, 2018-10-14:
No node in the GUI?  Did you check the sensor’s battery?  How many end devices are already connected to the RaspBee/ConBee?  What version of deCONZ and what firmware are you using?

Webserve, 2018-10-15:
i'm not using the GUI. Only PWA.

A few months back i paired this sensor. Last week i deleted it by accident but now i cannot pair it anymore. 

Battery is OK just to be sure i places a new battery. 
I got 20 other aqara / ikea sensors just working fine. 

Raspbee
Version 2.05.40 / 8-10-2018
Firmware 26240500

SebSemmi, 2018-10-15:
I have exactly the same setup.

ebaauw, 2018-10-15:
> i'm not using the GUI. Only PWA.
> your solution didn't work

You didn't try my solution.

Uakkauakkafo, 2018-10-21:
I have the same problem, the sensor water leak appear in the deCONZ as node 0x6666, but I'm not able to see it in the GUI of PHOSCOM and also I'm not able to find it with Postman application.
With Postman, I see all the sensor as REST object (and I can use in Home Assistant), the sensor water leak not appear.
here the screenshot of the deCONZ node

[image]

manup, 2018-10-22:
This doesn't look right, 0x1135 is the manufacturer code of dresden elektronik.

Did you pair it via Phoscon App and the solution described by @ebaauw ?

ebaauw, 2018-10-22:
Sharp.  Mine's got manufacturer code 0x1037, different values for the sizes, and definitely isn't a primary trust centre (see below).  The screen shot above looks like a combination of the RaspBee and leak sensor values.  Let's hope it's just the GUI that's confused.

[image]

manup, 2018-10-22:
I guess the wrong node descriptor is just a glitch in the deCONZ gui (I need to check how this could happen).

To refresh node descriptor or power descriptor manually &mdash; since deCONZ 2.05.42 the not so pretty buttons in the node config dropdown can be used to query the descriptors.

[image]

Note: it best works to press the button and then wake up the device within the next 7 seconds (afterwards most routers including RaspBee/ConBee will drop the message if the end device didn't poll).

ebaauw, 2018-10-22:
Cool, finally being able to read the power descriptor!  And the GUI displays the battery symbol after doing so!

manup, 2018-10-22:
Yep but be aware it's often implemented poorly, I did some tests which I'll put in the Wiki soon. Ikea end-devices for example claim that they are mains powered :)

ebaauw, 2018-10-22:
Looks like the battery symbol always shows full capacity?

manup, 2018-10-23:
Could be, the Climax sensors for example always show 66%, I don't really trust the power descriptor as battery state indicator, it might work for some devices, but we need more data to verify.

I think the power configuration cluster and IAS Zone state are the most reliable sources for battery state.

Uakkauakkafo, 2018-10-23:
So you mean that with the version 2.05.42 I should be able to view correctly the leak sensor?

eryckm, 2018-10-24:
I don't know if anyone here got the same issues but my sensors sometimes detect a leak around the same time and they keep showing as "wet" until I pick them up, put them under water and then dry them up. It takes a couple of weeks for that to happen. I updated to the latest version of ConBee (both adapter and app) to see if that will stop happening.

theGurk, 2018-10-24:
> Note that you need to squeeze the sensor quite hard for pressing the reset "button". It took me some time to figure that out. Even then, Xiaomi devices are notoriously difficult to pair.
>
> Try the [...]

Hi @ebaauw, 
I am running hass.io (HassOS) on a rpi and have some difficulties to add the leak sensor.
I had to change rpi due to the earlier was malfunction. I have a ConBee connected to the rpi but have trouble to add the leaksensors now. 

I have had all three added earlier but now I only get one to get connected. When I use Postman to check I can only find that one and not the other ones.

I am using the phoscon app and do not have the deCONZ GUI available.  Since I have the uniqueId for the other two I tried to POST-request them using Postman but only get an error message:

[
    {
        "error": {
            "address": "/sensors",
            "description": "Not allowed to create sensor type",
            "type": 501
        }
    }
]

My POST request looks like this:
{
    "config": {
		"battery": 100
        "on": true
        "reachable": true
		"temperature": 2500
    }
    "manufacturername": "LUMI"
    "modelid": "lumi.sensor_wleak.aq2"
    "name": "leak2"
    "swversion": "3000-0001"
	"state": {
            "lastupdated": "2018-10-23T20:51:51",
            "lowbattery": false,
            "tampered": false,
            "water": false
    },
    "type": "ZHAWater"
    "uniqueid": "00:15:8d:00:02:11:b2:a2-01-0500"
}

ConBee stick:
Version 2.05.39 / 22/09/2018
Firmware 26190500

Do you have any suggestions?

rsokolowski, 2018-10-25:
I have the same problem - can't add Xiaomi Leak Sensor. I already have 2 of them paired while ago (3rd went crazy reporting false leaks).
I've tried suggestion from @ebaauw but it didn't work.
I'm adding from PWA:
Vendor dresden elektronik
Product ConBee
Version 2.05.42 / 21.10.2018
Firmware 261F0500

The only relevant message that shows up in deconz logs during pairing process is:
[3.1] get IAS Zone type for 0x00158d00023a6415

Uakkauakkafo, 2018-10-25:
I tried your steps suggestion but i fail, i'm not able

bloody2k, 2018-10-30:
> I have the same problem - can't add Xiaomi Leak Sensor. I already have 2 of them paired while ago (3rd went crazy reporting false leaks).
> I've tried suggestion from @ebaauw but it didn't work.
> I'm [...]

I'm have exact same problem trying to pair my water sensor

20:52:26:882 [3.1] get IAS Zone type for 0x00158d0002349a7c

box99h, 2018-11-25:
Having the same problem getting the sensors up and running.
What I have for now is, that I can see two of the leakage sensors in the GUI. Also some of the values are read in the basic cluster. Can someone help?

[image]
[image]

ebaauw, 2018-11-25:
@theGurk, you can only add CLIP sensors through the API.  If the sensor was previously connected, you might be able to restore the resource by changing the deleted status in the database, if it’s still there.   Be sure to shutdown deCONZ and make a backup.

@box99h if the sensors show in the GUI, you’re almost there.  Just search for new devices in Phoscon, and re-read the _Basic_ cluster from the GUI, while pressing/squeezing the sensor to wake it.

@bloody2k , @eryckm , if the sensor is giving false reports, best check the battery.  These devices behave funny under low battery.

@all, it’s really tricky to get these paired, and without the deCONZ GUI you’re flying blind.  The REST API will only create the resource, when:
a) the network is open (through the API, so search from Phoscon or add sensor from the old web app);
b) it receives a message from the sensor; and
c) it has the data needed to recognise the sensor: the _Manufacturer ID_ from the _Node info_ panel, the mac address prefix, and the _Model Identifier_ from the _Basic_ cluster.

If you see the IAS zone message, you know the sensor has joined the network, but not whether the data has been read.  Without the GUI there’s no way to force reading the data, so your only hope is to reset the sensor and try again, hoping this time it remains awake and keeps polling its parent while the REST API tries to get the data.

closed 2019-04-01 by stale[bot]