dresden-elektronik/deconz-rest-plugin #830: Phoscon app design didn't show "new sensor" button on mobile phone

Counterfeiter, 2018-10-05:
Hi,

I was always using phoscon App with my iPhone SE (safari). I connected all sensors in the light menu with the button "add light". It works, but only sometimes.
I tought it is the way to go. As I displayed the phoscon app on the PC I saw for the first time the "add sensor" button.

[image]

Issue #827 is fixed if I use the correct button ;)

jjlawren, 2019-02-08:
I'm only able to see the button in landscape mode.

closed 2019-06-15 by stale[bot]

VoyteckPL, 2021-12-11:
> I'm only able to see the button in landscape mode.

That is really annoying. There should be a way to view this button in portrait mode