dresden-elektronik/deconz-rest-plugin #832: Unable to connect Aqara switch

flavio20002, 2018-10-05:
Hi, 
I'm not able to pair this switch

https://des.gbtcdn.com/uploads/pdm-desc-pic/Electronic/image/2017/06/06/20170606181922_10313.jpg

I already have 4 of this, and a bunch of other Xiaomi sensor. All paired without any problem. Just the last one that i bought doesn't pair. I've also tried to delete some of the old ones, reset them and paired successfully ad first try. Do you think my new is one defective? There is some trick or debug option, to understand what is wrong? Thanks in advance.

Phoscon App  2.05.31

flavio20002, 2018-10-05:
Hi, thank you very much for answering me. Sadly, I don't have deCONZ GUI, beacuse i don't have a monitor attached to my raspberry. Is there any other way, or maybe should I try to update to last version and see if it works? Thanks.

etique57, 2018-10-05:
You can connect to your pi using VNC Viewer.

ebaauw, 2018-10-05:
The commit I mentioned isn't in a release yet, so updating won't help you yet.

flavio20002, 2018-10-05:
I don't have any X server on my current Raspberry PI. I think I should by another Raspberry Pi and use it as developer machine. I should try to learn how to use deconz gui...Meanwhile, maybe the first release will solve my problem...

flavio20002, 2018-10-08:
I will wait the next release, because I'm not able to compile it myself. After that, I will update the issue.
Thanks to everybody.

zilluss, 2018-10-13:
Same here. I was able to connect it after compiling the REST plugin, but I only receive 1002 and 1004 events. The package on my switch said September 2018, so I guess it's the new version.

zilluss, 2018-10-16:
Long press working with changes from https://github.com/dresden-elektronik/deconz-rest-plugin/commit/fd1a7d0d6f8529b3a0499ef0293503185ac303c5

flavio20002, 2018-10-19:
Yes, now i get 1001 when pushed and 1003 when released after long press. The only issue is that is not shown in Phoscon App, but I want to check in the code