dresden-elektronik/deconz-rest-plugin #833: Safely delete junk data in the deCONZ database?

ACrazyConcept, 2018-10-06:
I have some old devices which keep coming up in the list and is unreachable even though I have deleted them multiple times via Phoscon.
Is there a safe way to get rid of that junk data in the database?

ronaldvans, 2018-10-08:
+4

manup, 2018-10-08:
The date is kept in the file `zll.db` this can be edited with tools like sqlitebrowser.

Normally devices should be deleted and stay so when they are deleted in the Phoscon App, I schedule some tests, seems there is somethings missing.

mariusmotea, 2018-10-09:
I have same issue with some IKEA lights used in the past for some tests. Every time i restart the raspberry the lights are added back and marked as unknow / unreachable. I need to manually delete them every time.

peer69, 2018-10-10:
Same here, I have some IKEA and OSRAM Light which I used to have, they magically come back every time deconz is restarted.

jo-me, 2018-10-11:
But even if they stay in state "deleted" there is a problem when repairing those devices again. They are not really re-added but just undeleted which makes it a pain if certain sensors were not discovered on the first paring attempt.

dem5867, 2018-11-28:
Can't seem to be able to find zll.db (Windows Based.).

manup, 2018-11-28:
It should be in the user home folder easiest is Windows Explorer file search there for zll.db

chimpy, 2019-02-01:
My experience with 58 and latest FW:

I had added and removed a Xiaomi round button switch in an earlier version:

[image]

It was not visible in the Phoscon webUI. I tried to add it again in 58 to no avail, even after changing the battery. So I opened the zll.db and deleted the offending record:

[image]

I then restarted deconz and tried to add the sensor. Added straight away:

[image]

Hope that helps! (All my other sensors seem to be still working with no problems after this intervention).

closed 2019-06-08 by stale[bot]

pepijnmm, 2020-03-08:
> Can't seem to be able to find zll.db (Windows Based.).
Found mine in: "C:\Users\My username\AppData\Local\dresden-elektronik\deCONZ" (took a while before windows found it so for the next person)

mat1990dj, 2020-04-07:
So If I understand correctly, somebody needs to manually delete the "deleted" devices from zll.db?
Or even doing so they will reappear?

pedolsky, 2021-09-09:
The problem is still exists.

MattL0, 2021-09-10:
Yup

BabaIsYou, 2023-01-18:
Seems still present #6665 (discovering it thru another issue)