dresden-elektronik/deconz-rest-plugin #834: Button editor feature, full brightness when already on

oscar-b, 2018-10-06:
New button editor work excellent, but it would be nice to be able to set it up so that it turns on when tapped, dims when held but also goes to 100% when tapped when already on. So two taps would always give full brightness, now you need to tap then hold and wait to get to full brightness.

manup, 2018-10-06:
It kinda is possible already by using a scene. Here is an example with the IKEA Tradfri remote.

* Short button press will call the scene
* The scene is configured to have full brightness, but could be anything
* Long button press will dimm up

[image]

Dim down could be achieved in the same way on the down button of the remote.

oscar-b, 2018-10-06:
Hm yeah but I like the fact that it turns on using the previous state as well :)

manup, 2018-10-06:
Ah ok, interesting idea, it would be possible by adding a specific rule. Might be a good default configuration as well. I'll put it on the internal discussion list. :)

oscar-b, 2018-10-06:
The “tap when already on” could be configured as “to 100%” or “next scene” perhaps, would work great for the small IKEA remote with two buttons.

closed 2019-02-10 by stale[bot]