dresden-elektronik/deconz-rest-plugin #835: Wrong coordinator after reboot

kurt4160, 2018-10-06:
I am using the latest version of deconz and the firmware. deconz runs in a docker container (debian). After a restart of the container the coordinator is missing in the network view - an osram wallplug gets the new coordinator.
Before reboot:

[image]

After reboot:

[image]

It also happens with a lot more nodes in the network - always the osram wallplug gets the coordinator after reboot.

flobernd, 2018-10-09:
I've got a similar problem with an `innr SP 120` (the only device in my network at the moment). It is recognized using the Phoscon App and can be controlled:
[image]

But after reboot, it just gets disconnected and marked as coordinator (blue):
[image]

I can manually fix this by clicking "reset node" (sometimes needs several tries), but after I reboot, it's broken again.

flobernd, 2018-10-09:
It is definitely a problem with this/these plug(s). Added a Hue bulb to my network and this one is recognized perfectly after restarting the service or the raspi.

kasteleman, 2018-10-09:
I have no problems with the innr sp 120. Never seen this happen in my environment. What version are you on?

flobernd, 2018-10-09:
Should be the latest version:
> Version 2.05.39 / 22.9.2018
> Firmware 26240500

Not sure if I did reset it correctly. Just pressed the button on it for some seconds until it started to blink. But as it got found in the Phoscon app after doing this, I think it should have worked.

flobernd, 2018-10-09:
Guess what ... I've just added a second plug of the same type and that one works like a charm. It even gets better: The defective plug started to work correctly as well. Don't know why ... Added and removed it several times before and even resetted the deCONZ bridge. Well ... at last it works now.

manup, 2018-10-10:
Confirmed, and fixed in 2.05.40

https://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.05.40-qt5.deb

This bug caused had more side effects, like some sensors and switches couldn't be integrated.