dresden-elektronik/deconz-rest-plugin #838: Deconz boot issue's after update

JMBokhorst, 2018-10-09:
Yesterday I noticed that HA had issue's communicating with the REST api. The state wasn't detected although deconz was running without any problem. After a quick look, I saw that there was an update for deconz and initialised an update for both the software and firmware of the conbee USB. 

However since I have done this, deconz doesn't want to boot normally anymore: 

Oct  9 11:39:58 boss-pi2 systemd[1]: Starting deCONZ: ZigBee gateway -- Initialisation...
Oct  9 11:39:58 boss-pi2 systemd[1]: Stopping deCONZ: ZigBee gateway -- REST API...
Oct  9 11:39:58 boss-pi2 systemd[1]: Started deCONZ: ZigBee gateway -- Initialisation.
Oct  9 11:39:59 boss-pi2 systemd[1]: deconz.service: Main process exited, code=exited, status=143/n/a
Oct  9 11:39:59 boss-pi2 systemd[1]: Stopped deCONZ: ZigBee gateway -- REST API.
Oct  9 11:39:59 boss-pi2 systemd[1]: deconz.service: Unit entered failed state.
Oct  9 11:39:59 boss-pi2 systemd[1]: deconz.service: Failed with result 'exit-code'.
Oct  9 11:39:59 boss-pi2 systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
Oct  9 11:39:59 boss-pi2 deCONZ[1791]: libpng warning: iCCP: known incorrect sRGB profile
Oct  9 11:40:07 boss-pi2 deCONZ-update2.sh[343]: Error: database is locked
Oct  9 11:40:08 boss-pi2 deCONZ[1791]: This plugin does not support propagateSizeHints()
Oct  9 11:40:08 boss-pi2 deCONZ[1791]: This plugin does not support propagateSizeHints()
Oct  9 11:40:14 boss-pi2 deCONZ[1791]: This plugin does not support propagateSizeHints()
Oct  9 11:40:37 boss-pi2 deCONZ-update2.sh[343]: Error: database is locked

I can make it work, by disabling deconz & deconz-wifi followed by removing / inserting the USB stick. Starting deconz afterwards results in the same errors, but I at least can connect to the webApp and control my lights. 

I'm running deconz headless (deconz-GUI disabled) & HA both directly on rasbian, and have followed the instructions of the main page to install deconz. Before yesterday I didn't have any issue's with this setup and also didn't change anything. 

Should be unrelated but besides deconz this pi also runs pi-hole. 

Does anybody has an idea how to solve this issue?  Currently I can't control my lights anymore

elmocito, 2019-03-11:
@JMBokhorst: So you can run deconz only stand-alone? I try to run it with a loxberry and have the same issue

JMBokhorst, 2019-03-11:
In the end, the issue for me was conflicting ports between the different services. Currently, I'm running everything in a docker without any issue's. I'm not familiar with loxberry services so I would check if there are any ports or overlapping dependicies. If it is possible you could also try running the deconz service in a docker.

elmocito, 2019-03-13:
Loxberry is using Port 80 and I changed that to 81. no additional ports.
I opened ticket https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1344 incl. error code.
No reply yet :-/