dresden-elektronik/deconz-rest-plugin #842: deCONZ headless installation with minimal Qt5 dependencies for Allwinner based BananaPi

moritzthecat, 2018-10-10:
I am somewhat stuck with the deCONZ installation on a BananaPi (Allwinner CPU) running Armbian because there are no Qt5 .deb installation files (and reluctant to compile full Qt5 though). The BananaPi is my preferred device as it comes with SATA interface having connected a cheap and robust hard drive. Since the BananaPi is used as a headless device in my use case and with no plans to install a desktop on it, I am thinking of the following scenario:

- install deCONZ as headless application with minimal Qt5 dependencies  on BananaPi, compile the required Qt5 dependencies but not the full Qt5 package 
- use a laptop or other device for the commissioning - then backup the database
https://github.com/dresden-elektronik/deCONZ-rest-plugin/issues/620
- restore device database on BananaPi and run the deCONZ headless REST API service as described in the user manual. https://github.com/dresden-elektronik/deconz-rest-plugin/issues/371.

There are some requests to de-couple the core from the GUI but it is stated here that Qt is needed for the REST API to work https://github.com/dresden-elektronik/deCONZ-rest-plugin/issues/9.

>  “…but there is already progress on this matter for some time to separate the core from the GUI” as reported in the above thread..

Questions is then if and how a headless deCONZ with minimal Qt dependencies could be installed on BananaPi ? Any thoughts and recommendations are highly appreciated certainly saving a lot of Qt5 compile time and disk space. Many thanx in advance.

MattL0, 2018-10-11:
An option would be a with docker i think : 

https://github.com/marthoc/docker-deconz

moritzthecat, 2019-02-08:
OK understand that Qt is deeply used in deCONZ although no GPU libs are necessary for headless use.
Will check the Ti route for the BananaPi (SATA) then - if time permits.
You may close this thread if no other idea.

manup, 2019-02-08:
>OK understand that Qt is deeply used in deCONZ although no GPU libs are necessary for headless use.

For historical reasons there are still parts in the core app which interact with widgets albeit these aren't shown. Over time we do refactor these to cut the dependency but there is no ETA when this process will be finished and Qt gui/widget dependencies can be dropped.

>You may close this thread if no other idea.

Closing the issue for now, it will take some time until minimal Qt5 is reached.

closed 2019-02-08 by manup