dresden-elektronik/deconz-rest-plugin #843: Serializing Conbee using ser2net and socat, possible?

cnrd, 2018-10-10:
Hi

As I have my server in the basement, I wanted to plug the Conbee into my Raspberry Pi (Which already hosts a Z-Wave USB), such that it was centrally placed in my home.

For the Z-Wave USB I use this config: https://community.openhab.org/t/share-z-wave-dongle-over-ip-usb-over-ip-using-ser2net-socat-guide/34895 but if i try to setup the Conbee in the same way, the deCONZ UI just shows an empty dropdown menu, as if there were no devices available.

I have tried exposing the Conbee as /dev/ttyUSB0 and /dev/ttyS0, but neither worked.

Are there any way to tell Conbee which port it should use for communicating with the Conbee?
Anyone who has this kind of setup working?

ebaauw, 2018-10-10:
Haven’t tried any usb to ip solutions.  Did try to connect the ConBee over a usb hub - failed, and over a 5m usb extention cable - works.

Why not run deCONZ on the pi?

cnrd, 2018-10-10:
I ended up doing that, it was mostly for speed, as a VM is much much faster to work with than my slow RPi2. The other thing is that I like to delegate tasks to devices, so I just wanted to pi to contain no configuration, other than the ser2net bridge, which takes about 5 seconds to setup. And then keep any deconz configs in a VM that get weekly offsite backups.

manup, 2018-10-11:
You may try the `--dev=/dev/ttyUSB0` parameter to specify the serial port device. Otherwise deONZ will try to find the ConBee via USB product/vendor Ids which might not be available in your setup.

cnrd, 2018-10-11:
Thanks, `--dev` does not seem to do anything when starting in GUI mode.

closed 2019-02-15 by stale[bot]

SAOPP, 2021-11-14:
Hey @cnrd from your last post, I didn't quite understand, did u manage to implement ser2net with the deconz stick, or did u just use a docker container on your rpi2?