dresden-elektronik/deconz-rest-plugin #845: Bitron 902010/23 4 Key Remote

Hypfer, 2018-10-10:
https://www.amazon.de/Bitron-Home-Fernbedienung-902010-23/dp/B00H3NIVLC

Not available via the REST interface. 

[image]
[image]

Hypfer, 2019-02-07:
This is still unsupported.

Hypfer, 2019-06-07:
Still relevant

joschaschultze, 2019-06-11:
I have two of them and would like to use this remote, too!

kcdr, 2019-09-24:
I have three of them. Would like to use them too.
Please try to support it in REST-API / Phoscon.
Thank you

Hypfer, 2020-01-22:
Still relevant

kcdr, 2020-01-22:
For me too.

nikolauzi11, 2020-03-07:
#2392

closed 2020-04-25 by Hypfer

ebaauw, 2020-06-08:
Bloody bot, it has already been closed. @Mimiix: config issue?

Mimiix, 2020-06-08:
> Bloody bot, it has already been closed. @Mimiix: config issue?

Bot did not close it. OP did.

ebaauw, 2020-06-08:
@Mimiix Then why is the bot is marking this closed issue as stale.

Mimiix, 2020-06-08:
@ebaauw Because its retarded? i dont know 😂