dresden-elektronik/deconz-rest-plugin #846: TRÅDFRI on/off switch false switching

Filialen, 2018-10-10:
Something is not right with the switch...
It's visible and responsive through the REST api, but not in Phoscon. And it turns on or off ALL lights in Phoscon, but only in the UI, not the lamps themselves!
Is there any developer that can verify this with an actual switch?

manup, 2018-10-10:
They will be visible and configurable in the Phoscon switch editor once they are available in Germany.

I guess (hope) the switch works like the OSRAM four button switch in its default configuration.

* broadcasts commands to all lights
* once bound to an group or lights, the switch will only send to this target

You may try to create a binding with the **Binding Dropbox** to an existing group identifier.

Filialen, 2018-10-10:
That's great! And they do work in Home Assistant, only fooling the UI for a minute when switched on or off.

Binding sounds promising, but I ain't able to bind. Timing out...
Binding on/off cluster both on switch and light. Any ideas?

[image]

manup, 2018-10-11:
After pressing Bind you likely need to press the switch a few times to keep it awake.

Filialen, 2018-10-11:
That did the trick! Now it did bind.
But, unfortunately, it keeps broadcasting commands to all lights in the UI...

It still works. Physically the switch operates just the right light. But both in Phoscon and HA, it turns on all the lights and then the software understands that the state of the other lights didn't change and revert back to the original state.

I guess this is the life in beta!

Filialen, 2018-10-11:
Yep! Creating a group in the light and binding that group worked perfectly!
All good now, no confused UI! :)

Thanks!

closed 2018-10-11 by Filialen

GatoPharaoh, 2018-10-27:
any advice on how what i could be doing worng
binding is reported as successful - but neither the device in the group is switched nor does it stop the remote from "fake toggeling" the other devices