dresden-elektronik/deconz-rest-plugin #848: Homeseer user. The deconz ip is always changing from local to standar ip.

MattL0, 2018-10-11:
Hi , here is the Homeseer log.  I do not think this is a homeseer plugin issues... so i post it here

`

ct-10 9:33:43 PM |   | JowiHue | Refresh bridge Phoscon-GW set to 30 seconds after establishing direct connection
-- | -- | -- | --

`

MattL0, 2018-10-11:
added   ExecStart=/usr/bin/deCONZ -platform minimal --http-listen=192.168.1.3 --http-port=80

to systemd service . Seems to do the tricks. I will report. I do not know why it can't stay on 127.0.0.1

MattL0, 2018-10-11:
Seems to be ok with 2.05.40?

Will report

manup, 2018-10-12:
I guess when the Websocket runs in a timeout after no events were delivered for some time, the plugin might close the connection?

wvuyk, 2018-10-12:
I don't think so from here. The plugin changes IP only based on UPNP calls. I have the impression that when deCONZ runs on the same machine as the plugin UPNP can return both local and network IP? It seems to occur on Linux mainly.

ebaauw, 2018-10-12:
> The plugin changes IP only based on UPNP calls.
> Seems to be ok with 2.05.40?

There was an issue that deCONZ would advertise the loopback IP address over UPnP, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/759#issuecomment-419644087.

MattL0, 2018-10-12:
I tried to remove upnp .maybe it was that..

There was an issue that deCONZ would advertise the loopback IP address over UPnP, see #759 (comment)<https://github.com/dresden-elektronik/deconz-rest-plugin/issues/759#issuecomment-419644087>.

MattL0, 2018-10-12:
So what should we do?

MattL0, 2018-10-16:
Up . 

Is disabling the upnp the only solution?

manup, 2018-10-16:
We need to check the upnp code, and maybe just stick to the eth0 interface IP address.

MattL0, 2018-10-17:
Just to be sure. Mine is enp4s0, not eth0

manup, 2018-10-18:
>Just to be sure. Mine is enp4s0, not eth0

My arch linux box also has a different ethernet name, but it should be simple to filter for the ip prefixes to detect the "right" one.

KodeCR, 2018-10-19:
The #817 pull request under discussion might help here, as it updates gwConfig["ipaddress"] once the network has initialised and that is used for the announce, while I believe the current code doesn't update this and just gets the ip-address each time it's needed for a rest-request without updating gwConfig["ipaddress"]/upnp.

closed 2019-02-23 by stale[bot]