dresden-elektronik/deconz-rest-plugin #850: Can I change the PAN ID and network key?

ebaauw, 2018-10-11:
I'm starting to think that maybe my 91-node network has become too complex (#849 , #806).  I'm considering to split it, i.e. connect half of my devices to a different deCONZ instance.  

Instead of having to pair half of my devices again, would the following approach be possible?
- Make a backup of the deCONZ configuration;
- Shutdown deCONZ;
- Restore the backup to a different deCONZ instance on a different Raspberry Pi and RaspBee, effectively cloning the installation;
- Delete half the devices from the cloned copy;
- Change the PAN ID and network key of the cloned copy;
- Delete the other half of the devices from the original installation.

From what I understand from the ZigBee documentation, it should be possible to change the PAN ID (on PAN ID conflicts) and the network key (when the message counter has been depleted), but I'm wondering if deCONZ supports this, and if the devices would be updated with the new values.

manup, 2018-10-12:
That's very tricky (impossible) to do.

I strongly recommend against running two setups based on the same backup in signal range.

Some details:
* Both coordinators would use the same mac address (from the original backup source).
* There can be only one pan coordinator for a given panid and extended panid, which is usually the mac address of the coordinator.
* The joined devices keep and use the extended panid in NVRAM. 

In short I don't think splitting the network without reset and rejoin is possible.
You may copy the `zll.db` to the other setup but make sure to delete/edit `zbconfig` table. 

At least for the lights changing networks it is pretty simple:

* Open network on the other setup
* Delete the light in Phoscon App, or via REST-API with `{ reset: true }`
* The light should leave and just rejoin the other setup

ebaauw, 2018-10-12:
> The joined devices keep and use the extended panid in NVRAM.

Thanks, I hadn't thought of that one.

closed 2018-10-12 by ebaauw