dresden-elektronik/deconz-rest-plugin #852: Want’s to install a destop for deconz

MattL0, 2018-10-12:
Hi , I would like to install a destop for deconz guy, on my ubuntu x64 server.

I tried one time and the phoscon app was creating a new network from nothing. 

I did disable the deconz service . And did enable the deconz service.

The deconz -guy was not present at boot. I had to click on it . 

Anyways. What would be the best procedure to install all of this with a light destop and the deconz , without messing up my current zigbeenetwork? 

Thanks

manup, 2018-10-12:
deCONZ uses the configuration in the file zll.db which is stored in the `$USER/.local/share/dresden-elektronik/deCONZ/zll.db`, where $USER is the one running deCONZ usually the one with uid 1000 as specified in deconz-gui.service.

To automatically start the deCONZ GUI you need to enable the respective systemd service.
**Important** A X11 must be running otherwise it won't work.

```
systemctl enable deconz-gui
```

MattL0, 2018-10-12:
ok will make a backup of my hdd. And will try some things

By x u mean xubuntu?

MattL0, 2018-10-12:
Ok was able to install. But reverted back

Maybe my cheap videacard  ( i have a amd2700x)  does not like destops. 

It was always freezing in the deconz gui.

Will wait to install on something else. Maybe will boot a raspberry pi with a ssd

closed 2018-10-13 by MattL0