dresden-elektronik/deconz-rest-plugin #854: Aqara endpoints don't react if they are further away

Jasonthefirst, 2018-10-12:
I am not quite sure if this is the right place but I'll try anyway. Maybe someone has experienced the same problem.
Deconz is running on a raspberry pi 2 in the newest version. I have a deconz addon for my RPI running the newest firmware and phoscon software. (2.05.40)
In my home there are running a lot of osram lights and a few osram plugs as zigbee repeaters. As endpoints I have a lot of aqara (xiaomi) sensors like a few magiccubes, doorsensors, temperaturesensors and motionsensors. 
It appears the xiaomi sensors won't route through any of the osram plugs and lights and as they get farther away from the bridge the system won't get updates or sometimes very late (especially the motion sensors from xiaomi/aqara). This is very frustrating as they should turn on the light in the bathroom but sometimes it takes forever until they send the motion. 
If I can clear anything up with some screenshots or more informations I would be happy to provide them.

ubatra, 2018-10-12:
I am seeing the same thing, Aqara sensors dropping off and not connecting through the plugs.

Counterfeiter, 2018-10-12:
I have the same setup. The sensors definitly route to the osram plugs! I can see it in the deConz GUI, but I can also see they use the path with the lowest hop counts. So I have lot of red and orange signal connections and sometimes a message is dropped. I am not sure if it is a problem with aqara, a lot of people seams to have problems with osram plugs as repeater...!

Jasonthefirst, 2018-10-15:
@mdescher  This seems to be more or less intended for reduced power consumption. 
[homeassistant discussion](https://community.home-assistant.io/t/xiaomi-human-body-motion-sensor-timeout/23398) at least that is what others discovered. 
But normally the motion detection is relatively fast if the sensor is in good reach of the deconz.

mdescher, 2018-10-15:
Okay, thanks, that's what I also already supposed. I found that there is a config parameter "duration" which is set to 60 and controls the duration of the motion-state in seconds. Setting this to 20 makes the sensor report a nomotion-event 20 seconds after the motion-event instead of 60 seconds. Unfortunately the next motion event will still not be sent earlier than 60 seconds from the previous one. I hoped for the parameter to control both, the reset and the reactivation period.

ebaauw, 2018-10-15:
> Setting this to 20 makes the sensor report a nomotion-event 20 seconds after the motion-event instead of 60 seconds. 

No, the sensor does _not_ report anything on no motion.  `config.duration` controls a timer by deCONZ to reset `state.presence` autonomously, or it would remain `true` forever.  It would seem the Xiaomi sensor needs 60 seconds to "recharge", before it can next detect motion and send an report.

There's other sensors that do send a report when motion is no longer detected, like the Hue motion sensor.  We use `config.delay` to configure the delay (on the sensor) after which it sends the no motion event.

mdescher, 2018-10-15:
Ah, thanks for the clarification. :-)

mdescher, 2018-10-15:
One more question. Is the timer started on each motion event, even if not yet expired?

Example: if I set the duration to 65 seconds and due to constant motion the sensor sends motion events every 60 to 64 second, will the state remain on "motion" all the time and no changes will be sent on the web socket as long as there is motion?

ebaauw, 2018-10-15:
Should be.

SwoopX, 2018-10-16:
I have quite a similar issue with one of my Xiaomi door sensors. After a couple of days, it gets kinda lost. First, I though it could be a range issue since it is really far away from the GW. So I placed an Osram Smartplug nearby but without any effect. Still lost and a button push doesn't do anything either...

manup, 2018-10-18:
Assumption: the root problem might be that Xiaomi devices sleep a lot longer when the ZigBee standard prefers it, so parent notes like OSRAM plug might kick them out their child table while the sensor is sleeping.

Philips lights used to have a very long living neighbor and child table (still are?) these might be better repeaters here.

SwoopX, 2018-10-18:
Hm, it doesn't seem so. I still have a green connection to the Osram plug. Still, as if it was dead and unplugging as just the effect that I don't get a connection at all. Another sensor even farer away is directly connected to the gateway in green...

mdescher, 2018-10-18:
I started my tests with an Aqara window sensor which I attached right next to a ZWave Fibaro sensor. This way I could confirm it worked flawlessly for several days. The motion sensors however seem to be rather unreliable. I have three of them at the moment. One for example simply did not respond a few hours ago. In the deCONZ GUI I had a recent "last seen" date but the Phoscon app did not show any motion events. After some time it worked again without having changed anything.

So some more questions:

1) Does the "last seen" timestamp mean that deCONZ did actually receive signals from the device at that time?

2) Regarding the window sensor: does send the current state about every hour via the PUSH API. Does this mean the sensor really sends the status or is it deCONZ REST API which resends the last known state?

manup, 2018-10-26:
The last seen refers to the last received commands timestamp, but I think also when neighbors claim to have seen the device.

The hourly report is sent by all Xiaomi devices, the API pushes only when something is received.

mdescher, 2018-10-26:
Great, thanks for the clarification.

Fury1SOG, 2018-12-14:
I have an Ikea TRÅDFRI remote and E27 bulb, plus several Xiaomi Door/Window and Temperature sensors in the same room. The ConBee gateway is about 6-7 meters away in a separate room. 

According to the deConz GUI, the TRÅDFRI remote uses the E27 bulb as a repeater, but all of the Xiaomi sensors prefer to directly connect to the ConBee gateway. The links are all green, so no problems, but I have not see any of my Xiaomi devices mesh with Ikea repeaters.

closed 2019-04-20 by stale[bot]