dresden-elektronik/deconz-rest-plugin #856: deCONZ 2.04.40 removes devicemembership attribute?

wvuyk, 2018-10-12:
It seems that with the release of deCONZ 2.05.40 the groups attribute "devicemembership" is removed. In my plugin I am using this attribut to determine if the group at hand is generated for a sensor. Most often these groups are "hidden" for users of my plugin as there is little use for these groups in Homeseer.

But now this attribute is not present anymore and the plugin has no other way to recognise these "sensor" groups. I would prefer to have the attribute available stil, or is there another way of recognise a group for sensors?

Wim

ebaauw, 2018-10-12:
They removed this attribute (and some others) from the API in https://github.com/dresden-elektronik/deconz-rest-plugin/commit/3a29b4390708aa8306496558fc183235c7a6e1ff.

>  I would prefer to have the attribute available stil

Indeed, I don't care for the other attributes, but `devicemembership` does seem like a real loss (even though I don't currently use it).  Maybe we should start a petition to bring it back? ;-)

> or is there another way of recognise a group for sensors?

You can get this info from the sensors `config.group` attributes, but then you'd have to scan all `/sensors` resources, building a groups map yourself.  Alternatively, you can specify `Accept: application/vnd.ddel.v1` in the HTTP header (this is what Phoscon and the old web app do), but then you'll get all other "private" attributes as well.

wvuyk, 2018-10-12:
I sure would like to petition to bring the devicemembership back in the standard REST API 👍 as I am actively using it in the plugin.  Or at least an indicator that a group was created for sensors e.g.: `type:"SensorGroup"`?
And as I do not really use others, could skip the adjusted HTTP header.

MattL0, 2018-10-12:
will revert back to 2.05.39, until this is fixed

manup, 2018-10-13:
Upps sorry, didn't know anybody other than the WebApp/Phoscon App does use the devicemembership attribute. Thanks @ebaauw for bringing it back.

Please keep in mind the attribute name might be a bit misleading, since it only represents "the group is owned by the device". With the *switch editor* we also control regular groups via rules, the related sensor ids are not part of the controlled groups `devicemembership` attribute.

wvuyk, 2018-10-18:
With 2.0.5.41 the issue is resoved. Thanks all!

closed 2018-10-18 by wvuyk