dresden-elektronik/deconz-rest-plugin #861: PWA WiFi settings screen issues on iPad

the-ress, 2018-10-14:
I have 2.05.40 and the wifi settings screen show escaped `&mdash;` and the layout has some issues.

It’s also weird it says “active”, but I never activated it.

[image]
[image]

manup, 2018-10-16:
Indeed that looks like portrait mode wasn't checked.
We are currently working on a complete rewrite of WiFi settings for deactivated, access point- and client modes, the ui should be fixed then.

>It’s also weird it says “active”, but I never activated it.

There is a humble detection of wifi/networking configurations which not controlled by Phoscon App, the active here only means there is "something" active or configured and the screen is a extra step to allow 
 Phoscon to overwrite it.

manup, 2019-01-17:
Closing this for now, new WiFi settings are available since 2.05.55.

closed 2019-01-17 by manup