dresden-elektronik/deconz-rest-plugin #865: Aqara wall switches dont report state when pushing phisical button

mat1990dj, 2018-10-15:
Since latest master (15 - oct - 2018) the aqara wall switches (without neutral) dont report status change on the api when changed via phisical button.
If I change via api I can see that the status is changed, but using phisical button not.

mat1990dj, 2018-10-16:
I can see a lot of entries for each of the switchs like the one below:
Node zombie state changed 0x00158d00024dca3e

manup, 2018-10-16:
Can you post a screenshot of a selected node from the *Node Info Panel*?

When I remember correctly the `lumi.ctrl_neutral1` switches advertise the receiverOnWhenIdle and mac capabilities wrong (@ebaauw ?) which causes some problems.

mat1990dj, 2018-10-16:
[image]
[image]

mat1990dj, 2018-10-16:
When pressing the button on and off I can see the following logs:

18:15:59:525 ZCL attribute report 0x00158D00024DC9D9 for cluster 0x0006, ep 0x02
18:15:59:525 Simple_Descr_req zdpSeq: 29 to 0x00158D00024DC9D9 endpoint 2

18:15:59:759 no button map for: lumi.ctrl_neutral1 ep: 0x04 cl: 0x0006 cmd: 0x0A pl[0]: 000
18:15:59:759 ZCL attribute report 0x00158D00024DC9D9 for cluster 0x0006, ep 0x04

18:16:00:026 rule: 2, resource /sensors : state/status id: 4 (cond: /sensors/4/state/status) not found

18:16:00:674 ZCL attribute report 0x00158D00024DC9D9 for cluster 0x0006, ep 0x02
18:16:00:674 Incr. ZDP retry count 2 on item 5

18:16:00:908 no button map for: lumi.ctrl_neutral1 ep: 0x04 cl: 0x0006 cmd: 0x0A pl[0]: 000
18:16:00:908 ZCL attribute report 0x00158D00024DC9D9 for cluster 0x0006, ep 0x04

ebaauw, 2018-10-16:
> When I remember correctly the `lumi.ctrl_neutral1` switches advertise the receiverOnWhenIdle and mac capabilities wrong (@ebaauw ?) which causes some problems.

My `lumi.ctrl_neutral2` switches report _Receiver On When Idle_ as false in the _Node Descriptor_.  They're mains powered, and I find the switches seem to be responsive all the time, like they don't sleep.  If haven't had any issues with my switches, since implementing the _Time_ server cluster on the coordinator, and setting RX on receiving the Xiaomi special attribute.

> If on the deconz GUI I request a Read attributes from the On/Off cluster, it updates the state and also to the rest API and the phoscon app, but If I don't press the read attributes it does not update automatically.
> When pressing the button on and off I can see the following logs

So the switch does send (and deCONZ does receive) the right reports: endpoint 0x02 for the `/lights` resource, and endpoint 0x04 for the ZHASwitch `/sensors` resource.  I don't understand how the state cannot be updated.

> Node zombie state changed 0x00158d00024dca3e

This would happen when deCONZ receives a report from the switch.  Question is why deCONZ marked the switch as zombie.

> 18:15:59:525 Simple_Descr_req zdpSeq: 29 to 0x00158D00024DC9D9 endpoint 2
> 18:16:00:674 Incr. ZDP retry count 2 on item 5

Not sure if these messages are related, but afaik deCONZ would only read the simple descriptor on pairing or on startup.  If it doesn't get a response for the ZDP request, it could indicate routing issues.  And it would explain why the switch is marked as zombie.

What happens after you power cycle or reset the RaspBee/ConBee?

> 18:16:00:026 rule: 2, resource /sensors : state/status id: 4 (cond: /sensors/4/state/status) not found

Could this indicate a database corruption?  Maybe the link between the node and the REST resources is lost?

mat1990dj, 2018-10-16:
Ok so I brought some debug logs maybe it can throw some light...

I restarted the whole raspberry and the log shows a join network waited until the switches appear with green link, and leave network

You are mentioning a routing issue? could all this be because I added the smartplug?

> Could this indicate a database corruption? Maybe the link between the node and the REST resources is lost?
How can I do a sort of sanitize to repair from an eventual corruption?

I can see a lot of Mac Poll (as stated on https://github.com/dresden-elektronik/deconz-rest-plugin/issues/871) and a lot of Node zombie state changed

```
18:53:03:591 Master: read param with arg 0x18
18:53:03:591 Master: read param with arg 0x13
18:53:03:591 Master: read param with arg 0x13
18:53:03:607 Device protocol version: 0x0108
18:53:03:655 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 1, node: 0x0000
18:53:03:690 Master: read param with arg 0x19
18:53:03:743 Current channel 15
18:53:03:866 Daylight now: goldenHour2, status: 180, daylight: 0, dark: 0
18:53:03:935 CTRL ANT_CTRL 0x03
18:53:04:819 device state timeout ignored in state 3
18:53:05:676 DEV config changed event
18:53:05:676 Master: read param with arg 0x18
18:53:05:676 Master: read param with arg 0x13
18:53:05:676 Master: read param with arg 0x13
18:53:05:677 Master: read param with arg 0x18
18:53:05:677 Master: read param with arg 0x13
18:53:05:677 Master: read param with arg 0x13
18:53:05:689 Device protocol version: 0x0108
18:53:05:743 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 1, node: 0x0000
18:53:05:780 Master: read param with arg 0x19
18:53:05:835 Current channel 15
18:53:05:839 Node 0x00158D00024DCA20 is known by 0 neighbors, last seen 12 s
18:53:05:876 CTRL ANT_CTRL 0x03
18:53:05:958 Device protocol version: 0x0108
18:53:06:003 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 1, node: 0x0000
18:53:06:037 Master: read param with arg 0x19
18:53:06:079 Mgmt_Lqi_req zdpSeq: 32 to 0x00212EFFFF022AE6 start index 0
18:53:06:090 Current channel 15
18:53:06:131 CTRL ANT_CTRL 0x03
18:53:06:222 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 1, node: 0x0000
18:53:06:319 poll node 00:15:8d:00:02:4d:ca:20-03
18:53:06:319 Idle timer triggered
18:53:06:319 Force read attributes for node Luces Parking
18:53:06:320 Force read attributes for SensorNode lumi.ctrl_neutral1 (3)
18:53:06:320 Node 0x00158D0001858A52 is known by 0 neighbors, last seen 0 s
18:53:06:320 rule: 2, resource /sensors : state/status id: 4 (cond: /sensors/4/state/status) not found
18:53:06:322 ZDP status = 0x00 -> SUCCESS
18:53:06:322 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
18:53:06:322 ZDP Mgmt_Lqi_rsp zdpSeq: 32 from 0x00212EFFFF022AE6 total: 0, startIndex: 0, listCount: 0
18:53:06:398 ZDP skip fetch 0x158D00024826A8, invalid last seen [3]
18:53:06:799 Node 0x00158D00024826A8 is known by 0 neighbors, last seen 0 s
18:53:07:279 Node 0x00158D0002321509 is known by 0 neighbors, last seen 51 s
18:53:07:320 rule: 2, resource /sensors : state/status id: 4 (cond: /sensors/4/state/status) not found
18:53:07:598 ZDP skip fetch 0x158D00024826A8, invalid last seen [3]
18:53:07:759 Node 0x00158D000251AFA1 is known by 0 neighbors, last seen 13 s
18:53:08:239 Node 0x00158D000232794D is known by 0 neighbors, last seen 0 s
18:53:08:318 Wait 2s till query finished
18:53:08:320 rule: 2, resource /sensors : state/status id: 4 (cond: /sensors/4/state/status) not found
18:53:08:547 MAC Poll 0x02 0xA77C
18:53:08:547 verify 0x00158d00025253dc is child node after 829140 s
18:53:08:548 Node zombie state changed 0x00158d00025253dc
18:53:08:564 verify neighbor status: APP_SUCCESS (0x00) 
18:53:08:594 DeviceAnnce of LightNode: 0x00158d00025253dc Permit Join: 0
18:53:08:595 DeviceAnnce of LightNode: 0x00158d00025253dc Permit Join: 0
18:53:08:595 DeviceAnnce of SensorNode: 0x00158D00025253DC [1]
18:53:08:595 DeviceAnnce of SensorNode: 0x00158D00025253DC [1]
18:53:08:595 device announce 0x00158D00025253DC (0xA77C) mac capabilities 0x84
18:53:08:595 set fast probe address to 0x00158D00025253DC (0xA77C)
[... 525 lines cut ...]
18:53:31:759 Node 0x00158D00024DC9D9 is known by 1 neighbors, last seen 1 s
18:53:31:787 MAC Poll 0x02 0x7A2D
18:53:31:787 Node zombie state changed 0x00158d00024dc9d9
18:53:32:036 MAC Poll 0x02 0x7CC7
18:53:32:036 Node zombie state changed 0x00158d000251afa1
18:53:32:239 Node 0x00158D00024DCA20 is known by 1 neighbors, last seen 0 s
18:53:32:318 Wait 1s till query finished
18:53:32:320 rule: 2, resource /sensors : state/status id: 4 (cond: /sensors/4/state/status) not found
18:53:32:375 MAC Poll 0x02 0xA77C
18:53:32:375 Node zombie state changed 0x00158d00025253dc
18:53:32:602 MAC Poll 0x02 0xCC5B
18:53:32:603 Node zombie state changed 0x00158d00024dca20
18:53:32:624 Master: read param with arg 0x18
18:53:32:624 Master: read param with arg 0x13
18:53:32:624 Master: read param with arg 0x13
18:53:32:634 Device protocol version: 0x0108
18:53:32:682 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 1, node: 0x0000
18:53:32:715 Master: read param with arg 0x19
18:53:32:767 Current channel 15
18:53:32:808 CTRL ANT_CTRL 0x03
18:53:33:818 Daylight now: goldenHour2, status: 180, daylight: 0, dark: 0
18:53:43:819 Daylight now: goldenHour2, status: 180, daylight: 0, dark: 0
18:53:53:818 Daylight now: goldenHour2, status: 180, daylight: 0, dark: 0
18:54:03:819 Daylight now: goldenHour2, status: 180, daylight: 0, dark: 0
```

mat1990dj, 2018-10-16:
Also when I plug in the aqara smartplug I can see a lot of

`19:03:49:254 ZDP skip fetch, node 0x158D00024826A8 total unconfirmed zdp request count 2 [2]
`

which corresponds to the smartplug mac address

manup, 2018-10-16:
I'll see to put up a fix of the receiverOnWhenIdle for the lumi.ctrl_neutral switches. The stack can get really confused if this bit isn't set correctly.

The on/off handler for the `lumi.ctrl_neutral1` on endpoint 4 doesn't seem to be reached or the ResourceItem doesn't get updated.

I'm ordering some Xiaomi devices including the above ones for further checking.

ebaauw, 2018-10-16:
> The on/off handler for the lumi.ctrl_neutral1 on endpoint 4 doesn't seem to be reached or the ResourceItem doesn't get updated.

There’s no buttonmap for these switches, the buttonevents are hard-coded.  Even so, ep 4 is for the ZHASwitch - not for the light resource.

manup, 2018-10-17:
The problem might be a missing simple descriptor for endpoint 4.
@mat1990dj can you please post a screenshot with the endpoint list of one of the nodes?

(click, right circle on the node)

mat1990dj, 2018-10-22:
Sorry not answering earlier, I was unable to access the pi from outside home

This is what you are refering to? It's from one of the switches
[image]
[image]
[image]

mat1990dj, 2018-11-05:
I'm still having this issue.
The state is not updated automatically when using phisical buttons. Can I help for debugging?
Thanks!

Smanar, 2018-11-05:
Just for information, I have a similar problem, an electrical switch without neutral, and by defaut It don't send update data (on websocket) when I swichtch it on or off, even if I use phoscon. I thought it was normal, because I have enabled it using binding. I have bind the device cluster on/off on the coordinator (Conbee), and now It work without problem.
It s not a good solution ?

mat1990dj, 2018-11-06:
In my case it's also a switch without neutral.
Can you explain me how do yo bind the cluster on the coordinator?
Also, as shown on the picture I have 5 different clusters named On/Off, which one did you bind?

Smanar, 2018-11-06:
I m not in the same case than you, I have only 1 endpoint, and it s a on/off light, so it's for that I m asking if I need to check the code or if it's normal. In my case the device is see as a classic light and in your case you have light + switch. So for me It's normal to use the bind fonction in my case.

Display the "bind dropbox" and using the drag and drop
cluster 0006 from device > source
the complete endpoint from coordinator > destination
Bind

In your case to test bind, I will take one of the 3 from switch endpoint (04 05 06), or all 3. But I don't know your device, you don't have only 2 switchs ? what is the 3th ?

mat1990dj, 2018-11-06:
Well I tried playing with the bindings as you said but I don't get it to work.
My switch is an electrical wall switch without neutral with one single button, and I dont understand what endpoints 03, 04, 05 and 06 are for.
02 on the cluster info for 0006 can turn on and off the switch, and if I use the read button I can see the state being updated on GUI and on the API when using physical button
03 on the cluster info for 0006 does not turn on of off, and when using the read button and pressing the phisical button I can see the state go fast from false-true-false. It does not update state on API
04 on the cluster info for 0006 does not turn on of off, and when using the read button and pressing the phisical button I can see the state go fast from false-true-false. It does not update state on API
05 on the cluster info for 0006 does not turn on of off, and when using the read button and pressing the phisical button I can see the state go fast from false-true-false. It does not update state on API
06 on the cluster info for 0006 does not turn on of off, and when using the read button and pressing the phisical button I can see the state go fast from false-true-false. It does not update state on API

Its very frustating because from the API I can't detect if the device is turned on or off and it worked in the past.

ebaauw, 2018-11-06:
The Xiaomi switches have the bindings (to the coordinator) and attribute reporting hard-coded, afaik you cannot change them.  From you earlier log, and from the changes you see in the deCONZ GUI, the communication between the switch and the RaspBee/ConBee works, so the bindings are OK.  It looks like the link between the REST API resources and the node is broken.  I'd try resetting, completely removing, and re-pairing the switch.  By completely I mean: delete the corresponding rows from the deCONZ database.

mat1990dj, 2018-11-06:
Well before doing that I decided to update the deconz with the latest stable and without dev package.

For my surprise the switch that was not sending status updates to the API, now sends them.

The caveat now is that from the API I can't turn on or off the switch, and one that is by it's side. All the others that I have at home that are the same model, work without problem.

So I went to the deCONZ gui and I saw that from there I can't control any of the switches its giving the following error:
[image]

So at the end I opened the DB and deleted one of the device, re join, reconfigure, etc... and then that device worked perfectly but one stopped working, I deleted that one and anotherone stopped working. So at the end I ended up deleting all my xiaomi switches and recreating them.

I'm troubled about what caused this... because it all started joining a xiaomi smartplug

closed 2019-03-13 by stale[bot]