dresden-elektronik/deconz-rest-plugin #866: deCONZ for Ubuntu on NUC not starting

Klagio, 2018-10-15:
I have a NUC with Ubuntu 16.04 on it, I installed from terminal, and apparently no errors. When in the Ubuntu GUI, I click on deCONZ app and nothing happens

user@ubuntu:~$ sudo dpkg -i deconz-2.05.20-qt5.deb
Selecting previously unselected package deconz.
(Reading database ... 437781 files and directories currently installed.)
Preparing to unpack deconz-2.05.20-qt5.deb ...
Unpacking deconz (2.05.20) ...
Setting up deconz (2.05.20) ...
Processing triggers for desktop-file-utils (0.22-1ubuntu5.2) ...
Processing triggers for bamfdaemon (0.5.3~bzr0+16.04.20180209-0ubuntu1) ...
Rebuilding /usr/share/applications/bamf-2.index...
Processing triggers for gnome-menus (3.13.3-6ubuntu3.1) ...
Processing triggers for mime-support (3.59ubuntu1) ...

MattL0, 2018-10-15:
I am starting to thrink that the deconz gui has issues on ubuntu..

Mine was frozing and was not starting automatically

Klagio, 2018-10-15:
Anyone got it working?

Klagio, 2018-10-16:
Finally I managed to have the Ubuntu-GUI working. Strangely the GUI works and identifies 2 lights (a Hue bulb and a Xiaomi power plug), the webapp also identifies and controls them, although the phosom app does not: also integration in Home Assistant is not working.

WHat I did was install some dependencies by uninstalling mariadb, mysql, and reinstall from scractch mysql. Not sure what was what, but now the GUI works

closed 2018-10-16 by Klagio