dresden-elektronik/deconz-rest-plugin #868: GE Zigbee Appliance Control Switch (45853GE) and Smart Dimmer Switch (45857GE)

wayneflo, 2018-10-15:
I registered a new conbee (latest firmware), put the GE Appliance Control into pairing mode (immediately after a reset), then used the deconz web app to "open the network," then "scan for devices."

I also put the GE Appliance Control into pairing mode (immediately after a reset), then used the HS3Pro jowihue config page to "scan new devices."

Neither scan recognized the GE Appliance Control.

I have  not tried my Smart Dimmer Switch, but, since it is not mentioned on your supported devices page, I assume I will get the same result.  

I have a house full of the Zwave versions of the GE products, and like them a lot.  I would appreciate you adding support for the Zigbee versions.

ebaauw, 2018-10-21:
The GE in-wall dimmer has almost the same fingerprint, see #891.

wayneflo, 2018-10-22:
I suspect the GE in-wall smart switch, 45857GE, is also the same footprint.  I don't have one to post the info.

closed 2019-05-23 by stale[bot]