dresden-elektronik/deconz-rest-plugin #871: Question: What is the meaning of colours of upper left circle?

mat1990dj, 2018-10-16:
I'm having some devices not updating status, since I have this issue I can see that the devices with the issue have the upper left icon always flashing in green, what does it mean? can I stop it?

[image]

manup, 2018-10-16:
Which modelid has the device?

Green is fairly new and means the device send a mac data requests (poll) to the gateway, that's not an error, it just wasn't visible in older deCONZ versions.

mat1990dj, 2018-10-16:
It's one of the aqara wall switches without neutral (lumi.ctrl_neutral1)

I also tried going back to 2_05_38 and I still got it. The thing is that since I have this, those devices are not automatically updating the state of the On/Off cluster (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/865)