dresden-elektronik/deconz-rest-plugin #872: Raspbee configuration reseted

mat1990dj, 2018-10-16:
After investigating some issues I've found that my configuration inside deconz seems invalid:

[image]

How can I get the default values? or previous values?

manup, 2018-10-16:
Doesn't look like deCONZ has retreived the configuration at all.
Is the RaspBee connected and visible as node?

mat1990dj, 2018-10-16:
Yes... :(

[image]

manup, 2018-10-16:
That's weird, since there seems to be a active network and coordinator has valid mac address.
So when you press *Read* or restart deCONZ is still stays all zero?

mat1990dj, 2018-10-16:
Yes
I'm thinking I'll have to reset my gateway

manup, 2018-10-16:
I wouldn't do that. Please first check the zll.db file with the sqlitebrowser and look in the table `zbconf` here all prior configurations are stored.

mat1990dj, 2018-10-16:
Ok, so I got the configuration from that table and put it into the deconz gui, still missing TC Link Key, can this missconfiguration be the source of various problems?

{"apsUseExtPanId":"0x0",
"curChannel":15,
"deviceType":0,
"extPanId":"0x212effff022ae6",
"fwversion":"0x26240500",
"macAddress":"0x212effff022ae6",
"networkKey":"00f4d0bcfc24f40064b4946440d0d8d8",
"nwkAddress":"0x0",
"nwkUpdateId":0,
"panId":"0xe4be",
"securityMode":3,
"staticNwkAddress":false,
"swversion":"2.05.40",
"tcAddress":"0x212effff022ae6"}

[image]

manup, 2018-10-17:
The TC Link Key is normally:

`0x5a6967426565416c6c69616e63653039`

This is the default HA trust center link key (ZigBeeAlliance09) and is used when new devices join the network.

Note in order to apply network settings changes you need to Save > Leave > Join in deCONZ.

closed 2019-02-21 by stale[bot]