dresden-elektronik/deconz-rest-plugin #874: Busch Jaeger wall switch top butttons

widert, 2018-10-16:
I have some 4 gang Busch Jaeger switches that is used without any dimmer or switch "connceted" to the top row buttons (it is just connected to 5v) . However I am not able to assign anything to the two top buttons?

ErnstEeldert, 2018-10-17:
As far as I understood the manual that comes with that switch, those buttons are hardwired to the switch/dimmer itself.

widert, 2018-10-17:
Iḿ sorry but I belive you are wrong, ref https://www.busch-jaeger.de/en/products/product-solutions/remote-control/zigbee-light-link/

There are 3 options for the top row buttons, switch, dimmer and "radio", i.e. the button is connected to a 5v power supply only. The button should in many aspects work the same way as the battery operated ones, except that it is hardwired to 5v power supply.

ErnstEeldert, 2018-10-17:
Ah, I think I misread the question. I have a dimmer unit attached to the 4 way control, and although I can assign lights to the top buttons, these settings are ignore. It's just the dimmer that's being controlled by the top buttons.

widert, 2018-10-17:
I cannot assign anything to the top buttons, they are unfortunately greyed out :(

manup, 2018-10-17:
It depends on the switch, the BJ switches which control a relay or dimmer have the top row reserved to control the wired light by these buttons, they aren't available for ZigBee commands.

[image]

So it depends on which one you have `01`, `02` or `03`?

manup, 2018-10-17:
Interesting in that case the top row should work, I'll see to verify the code.

widert, 2018-10-25:
> Interesting in that case the top row should work, I'll see to verify the code.

Did you have a chance to look at this?

manup, 2018-10-28:
Hello not yet, had to figure out some routing/config problems first, but it's on the todo list for the next version.

widert, 2018-10-29:
Ok, thanks

weakfl, 2018-11-01:
> It depends on the switch, the BJ switches which control a relay or dimmer have the top row reserved to control the wired light by these buttons, they aren't available for ZigBee commands.

@manup Sorry for sidetracking this issue, I hope you don't mind me asking a related question:
Are the switches connected to a dimmer/relay supposed to send a report/event when they change state?

I'm currently using the openHAB zigbee binding and the switches are configured to send reports on state change and every two hours. However, they're only sending the two-hourly reports but nothing if the top rocker is pressed (the relay/dimmer changes state). That's kind of problematic because the relay state is not immediately reflected in whatever software UI is used.

**How does deconz handle this?**

manup, 2018-11-02:
Status update on Busch-Jaeger switches: Beside ui inconsistencies there seems to be a problem with the group/binding creation to enable capturing of events. 

>@manup Sorry for sidetracking this issue, I hope you don't mind me asking a related question:
Are the switches connected to a dimmer/relay supposed to send a report/event when they change state?

The top buttons (only dimmer and relay) just control the load by physical press, they don't send any events. Only the related light resource may give a hint of activation.

weakfl, 2018-11-03:
> The top buttons (only dimmer and relay) just control the load by physical press, they don't send any events. Only the related light resource may give a hint of activation.

Thanks for the confirmation @manup, I was afraid that's the case. I have written BJ a couple of times and asked them to address this issue, but it doesn't look like they're willing to fix it. It's a shame, apart from the UI issues the relays/dimmers are pretty good.

widert, 2018-12-08:
I’m currently using the Hue hub and can assign all buttons as expected. Including the top row.

closed 2019-04-14 by stale[bot]